openhardwaremonitor/openhardwaremonitor #1242: CPU temps stopped showing

SweepsMT, 2020-03-27:
This is a friends PC. His temps were showing fine but recently just stopped showing.

[image]

DarthDestroyer, 2020-04-17:
Could you add a Report?
Was it started as admin?

SweepsMT, 2020-04-19:
[OpenHardwareMonitor.Report (1).txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4499989/OpenHardwareMonitor.Report.1.txt)

SweepsMT, 2020-04-19:
and yes was started as admin

MerlinStWahgwaan, 2020-04-26:
I am having the same issue now as well... i had all readings fine yesterday, now misssing Core Feq and Temps for CPU, but still getting Load values....

Intel XTU, Task Manager, and other programs can read CPU values fine still, not data loss on them, just on OHM and only today, worked fine yesterday and every day for the last 3 weeks that I've has it....

MerlinStWahgwaan, 2020-04-27:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4536757/OpenHardwareMonitor.Report.txt)

MerlinStWahgwaan, 2020-04-27:
ive noticed its not pulling correct system info either, for instance, im not using Windows NT, im on windows 10 Pro

SweepsMT, 2020-04-27:
i've noticed this error on both my report and yours now that im reading it

Ring0

Status: Installing driver "C:\Users\Benhimovic\Desktop\Tools\OpenHardwareMonitor\OpenHardwareMonitorLib.sys" failed and file exists
First Exception: Service already exists
Second Exception: StartService returned the error: Windows cannot verify the digital signature for this file. A recent hardware or software change might have installed a file that is signed incorrectly or damaged, or that might be malicious software from an unknown source. (Exception from HRESULT: 0x80070241)

MerlinStWahgwaan, 2020-04-27:
yes ive noticed that as well, not sure if it means anything.... I cant figure out if theres a windows service thats not running or what, i know the programs creats the lib.sys file when it tries to launch, but then it disappers and the program shows what you see....

I cant explain how it works one day and not the next.... 

I THINK its no able to read the data from resource monitor correctly, but again, not idea why or how it could become broken while my laptop was off all night, and it was working yeasterday forsure....

MerlinStWahgwaan, 2020-04-27:
I GOT MINE BACK

I CANT BELIEVE IT......
[OpenHardwareMonitor.Report2.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4537091/OpenHardwareMonitor.Report2.txt)

Its still not showing the correct OS in the report, but im getting my sensor readings, and they match my opther software, XTU, AWCC, Core Temp, ect....

Before I post what I did, let me know, your in windows 10 right?

MerlinStWahgwaan, 2020-04-27:
So i began back tracking what I was doing last night and what i was playing with in my laptop that could have caused somthing like this, and I remembered I had been turing on Virtualization and Windows Sandbox so I could use that for testing in a secure environment... anyways....

Thats when I decided to basically undo everything I did last night and see if by chance it solved my issue, and it did.

This is exactly what I did, and I assume it will work for you if you do the same thing; As long as your not using Hyper-V for anything other than Sandbox, which apparently doesnt require any of that running.

Open the Start Menu, [WIN] key, and start typing "Windows Features"
Open "Turn Windows Features On or Off"
Turn off the following (remove the check boxes)

- Guarded Host
- Hyper-V
- Virtual Machine Platform
- Windows Hypervisor Platform

Once I turned those off, and rebooted as required, my OHM worked immediately.

Im not sure which of them is causing OHM to not be able to read the data from the monitor chips, but its obviously somthing to do with Hyper-V being enabled.

Hopfully someone with more knowledge than myself can track down what the issue is and get a proper fix so it can run with hyper-V enabled.

SweepsMT, 2020-04-27:
ok awesome thanks! i'll try this on my friends pc via teamviewer soon and see if i get anywhere.

Also I think it's not wrong in saying Windows NT, there's some technicality's i'm not going to explain lol.

MerlinStWahgwaan, 2020-04-27:
Cool, let me know if it works and fixes the readings for your friend, it would be nice to have a verified fix so others know forsure it can fix the problem, I know I hate threads with only 1 "possible" fix and they just go dead with no one saying if it works for them or not, lol.... Would also be good to verify the new issue i started referencing this thread about the Hyper-V causing the data loss....

SweepsMT, 2020-04-27:
this was not the fix for us however i tried a few things in bios and it worked

I turned on win 10 WHQL that wasn't enabled for some reason, and disabled virtualisation tech. not sure if both or which one fixed it.

MerlinStWahgwaan, 2020-04-27:
ooh shooot... he was running in Legacy Boot not UEFI secure boot? wow, i didnt even consider that not running in UEFI boot could stop the data from the sensors, but i guess they require some kind of secure driver.... 
was that computer upgraded from Windows 7/8 to windows 10? 

that would be the only way I can think of that would cause windows 10 to be running in Legacy not UEFI secure boot, becuase I was under the impression all computers sold with windows 10 were installed using the UEFI bios not legacy.....

And you were able to enable it and boot the computer without any issues?
WOW again, lol I would think changing over to secure boot UEFI would have caused atleast some errors when verifying the drivers of 3rd party programs,m if not caused a completed re-instalation of windows and partitions on the drive.

Anyways, well now we know atleast 2 things that can cause OHM too loose CPU data.

garyfritzz, 2020-05-18:
I have the same problem, no temps or clocks.  Running Win10 Pro on HP Spectre x360 15 with i7-8705G CPU.  I don't have any of the Windows virtualization options turned on, nor am I using legacy boot.  Any ideas?
[image]
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4641560/OpenHardwareMonitor.Report.txt)

MerlinStWahgwaan, 2020-05-18:
Have you tried using Core Temp Instead?
Does it show any Temps?

From you screenshot it loos like that comp isn't providing Temps for
some reason, I may be wrong, but when I was having issues, I could still
see the categories and titles, I just didn't have the data loading... since
you don't show the temp titles as all, im curious if that comp is just
monitoring Temps for some reason...

What about other programs? Like Intel XTU? Is it showing CPU temps?

I just want to be sure its actually OHM not reading the temps....

If it is for sure OHM, try moving the EXE to a higher folder tree, like
move it to your desktop directly and run it as an administrator.. try both
of those two things and see if they help...

Some times programs don't like long location directories... if moving the
exe location and running in the new location as administrator doesn't help,
gotta keep trying more thing, but let's start with all this first...

garyfritzz, 2020-05-18:
Thanks, I hadn't tried Core Temp.  It works (see below), as does CPUID HWMonitor.  It definitely seems to be OHM.

I had it installed in C:\Program Files (x86)\Utils\OpenHardwareMonitor.  I tried copying everything to C:\, and no change.

[image]

MerlinStWahgwaan, 2020-05-18:
Hmmmm...... thats a strange one

If you see a few posts above, I posted my OHM report when I was missing
temp info, and if you open it, youll see the titles are still listed even
when the sensors were not detected/reading..

Its very strange your report doesn't show the temp sensors without the
data....

I'm not sure what could possibly be wrong, I assume your running as admin,
and that would be the only other thing I know of that could stop it from
working....

Only other thing I can suggest, is restarting into safe mode or even
better, the windows admin account, and try it in those two places... if the
temps show in either of those two cases, its somthing to do with a program
your running or your Microsoft account settings....

Its doubtful, but maybe your firewall or anti-virus is some how stopping
it? Not sure...
Safe mode or the default Admin account would be the way to find out... if
it still doesn't work in safe mode or the admin account, I honestly have no
other ideas for you... i don't know of any process or service that needs to
be running for the temps to show, and even if there is, Core Temp is
working so obviously thats not the issue....

Not too sure.. keep us posted with what you try, hopefully someone else has
some ideas for you... if you have any experience, you can download the
github repo and recompile your own OHM using Visual Stuido. But thats a bit
more advanced, don't attempt it if your unfamiliar with the process...

MerlinStWahgwaan, 2020-05-18:
You could also double check your BIOS settings, look for and see if you
have a setting for secure boot or WHQL, if its disabled, try turning it
on... you may be on UEFI boot not legacy, but it may need secure boot
running too, like the other people said before as well, he wasn't sure if
WHQL on or virtualization off fixed his problem because he did both at the
same time....

garyfritzz, 2020-05-18:
Yes, I did run it as admin.  No difference.

I'm a bit swamped to try the Safe Mode, BIOS settings, etc.  Maybe later.  In the meantime I can use Core Temp to watch my temps.  Thanks for the pointers!

MerlinStWahgwaan, 2020-05-18:
No problems!!

Let me know what happens with the BIOS and safe mode when you do get time,
always good to know what does and doesn't work

I'll keep my fingers crossed and keep you posted if I think of anything
else that may be of use to your troubleshooting, unfortunately short of
remote access its hard too know forsure what could be causing the issue
with your particular system... just check the BIOS, try safe mode, make
sure HyperV and the stuff i listed in the posts above is disabled, and let
me know from there....

Have a good night, and good luck!