openhardwaremonitor/openhardwaremonitor #1244: Expanded/collapsed sensors not remembered since 0.9 branch

GastonWazeef, 2020-04-04:
Hi,
I was using 0.8.0 Beta for a long time and very satisfied with it.
However since I upgraded to 0.9.2.0, the app doesn't remember the collapsed sensors and expand all of them at each launch... Could you please allow the app to remember last display as before?
Thanks!
[image]

moellerm, 2020-05-05:
This has never been a feature in the official version of the OHM. But I have implemented this in commit 87c6f02. You can test it in the just released OHM version 0.9.3

closed 2020-05-05 by moellerm

GastonWazeef, 2020-05-05:
Wow! I didn't expect such a fast patch! It works perfectly, thanks a lot!!!