openhardwaremonitor/openhardwaremonitor #1246: Using Gadget view during gaming

RomanIvasyk, 2020-04-07:
Hello. Is it possible to use the gadget view while playing? If yes - please tell me how. If not - I think it would be cool to add that feature

TheTrueBumba, 2020-04-10:
You are asking about gaming overlay. Forget it. Its another beast and need to be made with support of anticheat creators to avoid false positives and bans.
But I have a second monitor and in some games the gadjet disapears in the second one. Ie: Works fine with GTA but disapear with ARK. I am testing Sidebar Diagnostics, made using openhardware monitor and works like a charm, but show only a few sensors.

MerlinStWahgwaan, 2020-04-27:
if you really want it, you can put your game into "Full Screen Windowed Mode" and it will overlay.... in my game, Star Citizen, I can just press Alt + Enter at the main menu and it goes into that mode and the overlay works..... i think some games it needs to be done in other ways, steam i think has settings for it....

either way, you can have it work, you just have to pop the game out of full screen mode as full screen games work differently in Windows 10, they dont allow overlays that aren't "hooked" into the running game, thats how MSI Afterburner and RiviaTuner work....

closed 2020-05-02 by RomanIvasyk