openhardwaremonitor/openhardwaremonitor #1247: ASRock X570 Phantom Gaming-ITX/TB3 no temps

HanzoX7, 2020-04-10:
Hi! Been using this great app a few days now and love the gadget! Just noticed the motherboard section doesn't have any + signs to expand and see temps and other data.
[image]
Attached a report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4462948/OpenHardwareMonitor.Report.txt)
Hope this is an easy fix...

DarthDestroyer, 2020-04-17:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xC732 at 0x2E/0x2F

Your Chipset is not supported, yet...

MerlinStWahgwaan, 2020-04-27:
your mother board wont expand, but your processor and stuff will and should show temps.... mine doesnt either, limited support, i dont even know what sensors would be for the motherboard itself....