openhardwaremonitor/openhardwaremonitor #1251: in-game hardware monitoring overlay text

aha999, 2020-04-17:
It would be awesome if you could have an in-game / over the screen overlay text of whichever data we set, like Temperatures for example, etc. So we could use it to monitor cpu/gpu temperatures, voltages, fan speed etc. while gaming or when doing anything. It would be really useful

DarthDestroyer, 2020-04-17:
The Idea is not new, but it will be difficult!
The Problem is, that you have to interact with the used graphical api (DirectX, OpenGL, or whatever) to "overdraw". 
Its not impossible, but in some Games it will be detected as cheating.

DarthDestroyer, 2020-04-17:
Same "Issue" #1246

javguit, 2020-04-20:
This can be done by adding RTSS API to the program. Cons are that you need the RTSS Server running for it to work.

DevTechProfile, 2020-04-21:
CapFrameX combines OpenHardwareMonitor and RTSS.

https://github.com/DevTechProfile/CapFrameX

MerlinStWahgwaan, 2020-04-27:
put the game in "Full Screen Windowed Mode", for Star Citizen i press Alt + Enter at the main menu, and it pops into that mode and the gadet works fine as an overlay then

MerlinStWahgwaan, 2020-04-27:
if you really want it, you can put your game into "Full Screen Windowed Mode" and it will overlay.... in my game, Star Citizen, I can just press Alt + Enter at the main menu and it goes into that mode and the overlay works..... i think some games it needs to be done in other ways, steam i think has settings for it....

either way, you can have it work, you just have to pop the game out of full screen mode as full screen games work differently in Windows 10, they dont allow overlays that aren't "hooked" into the running game, thats how MSI Afterburner and RiviaTuner work....

javguit, 2020-05-04:
I found a way to add this, but i need OHM to be compiled in at least 4.0. Is there a plan to upgrade the framework?

MerlinStWahgwaan, 2020-05-04:
bahaha, doubtful budd.... and I dont think anyone want to use RTSS with OHM, if they are using OHM they most likely dont want or need RTSS unless they are also using Afterburner, but not all ppl use that. its best to keep OHM as it is with current gadget, and just pop your game out of full screen into windowed mode, and their gadget stays on top with no alteration of the software or additional programs like RTSS

javguit, 2020-05-04:
Well that's just too bad. My goal doing this is to improve OHM, as I'm sure is yours too. I don't see this as something negative, on the contrary. Me myself use this a lot and its a great way to have all the info you want on the display while you stress test or game or whatever it is people use this for.
RTSS provides a way to show all the data you want OSD, so that program can handle all the 3D stuff (or whatever it is it handles), and OHM only handles the info.
If you don't care about OSD, that's fine, nothing will change for you. Having RTSS installed with the addition I'm proposing is not necessary if you don't want to use OSD data.

MerlinStWahgwaan, 2020-05-04:
Ok, so, I'm not in the mood today, but I'll be quick and I'm done on this
thread because it's stupid and already been answered.

First, this is the github issues page, not feature suggestion page or
thread, this thread was to answer the original posters issue, if it was
currently possible with OHM as it stands, nothing else. So go start a new
thread if you want to discuss features being added. Unlike you, I was
trying to be constructive and helped find a way to use the existing program
to do what was requested without wasting devs time and using additional
software people might not trust or want installed like RTSS which you need
to leave running, further degrading overall cpu performance because of the
ram and cpu loss to that software, I loose 1.0MB of my ram to RTSS
processes not even running a game or overlay, just sitting on windows
desktop doing nothing, jeeze Cortana uses less, only 0.4MB, and it's always
listening for her trigger.

Second, your suggestion totally negates the entire purpose of OHM, that it
is a standalone, no install, program that does not require any additional
software of any kind, and includes it's own gadget OSD for stuff like
stress testing or while playing games.

Third, if your going to act like adding in a feature that litterally maybe
5 people might want since your the first ever to suggest it, i checked the
threads, and that its a good idea and that wasting the developers time on
your request is fine because you want it, and think it's a good idea, you
might want to remember that OHM is freeware and runs on donations, it's not
Microsoft charging $100 per copy, nor using ads or anything else to
generate revenue.

So if you want somthing added, that this program is purposely and
specifically designed to not need or use, and that Windows 10 specfically
changed for full screen apps to not allow overlays not connected to the
running app, or "hooked" into the app, added to it, you should try making a
proper thread, or even better, a donation to the devs, instead of telling
them you want them to upgrade their base software to accommodate your wants
that they specifically designed this program to avoid.

There, I'm done with this. It's an issues page not suggestions or arguments
or nonsense page.

Stay safe during COVID, have a nice life  🖖

javguit, 2020-05-04:
I'm telling you I already have a version working, but need the OHM framework upgraded to 4.0. The image i uploaded was running under LHM, fork of OHM compiled in 4.0.

MerlinStWahgwaan, 2020-05-04:
If you have done so already, and you had it working, as per your screen
shot, why does the firmware need to be changed to 4.0 language?

Also, if that's the case, go start a new thread, name is somthing like;

Feature Request: work with RivaTuner Statistics Server OSD

And then post what you can, your screenshot again, what you did to get that
far, ect... the more information and detail, the better so I understand
what's going on.

I will come there and see if there is anything I can do to assist you,
because I already have the full github repo for OHM, and visual studio, and
I've already updated the programming language for OHM in VS to 7.3 and
built the program when I was having an issue with no cpu frequency and
thought it was a problem with OHM, but it ended up being somthing to do
with Hyper-V....

But bottom line is, I might be able to assist with a custom build/rebuild
of OHM for you if that's what you need right now..... but I dont think the
devs will ever attempt this themselves, I'm surprised you did, making the
two programs communicate correctly in real time is not so easy, never mind
interacting with the Gapi and overdraw crap....

javguit, 2020-05-04:
I found a Github repo (https://github.com/spencerhakim/RTSSSharedMemoryNET) that is based on the RTSSSharedMemorySample provided by RTSS that provides .net support, but it's compiled in 4.0 and cant add a reference to that dll in OHM because of the framework version incompatibility.

avengerx, 2020-07-01:
I am sure this is not a solution for this problem but, in my case, I have two screens, I basically have the second screen for monitoring purposes (and for terminals as I basically work programming stuff). So I just always leave the gadget in the secondary display so while playing games at full screen I can see what's up on temps and other things.

I agree it is a very useful thing to have overlay support for OHM so you can keep an eye on stuff while playing heavy games (that you may want to make sure that new water cooler solution is keeping everything cool), but also agree that OHM as a standalone simple tool is great.

I believe the mid-term between the two would be the ability to allow external programs to "read gadget data" from OHM so a third party overlay tool could easily pull the values off OHM without the need to change the whole concept. So you could check a box in OHM and it would allow other applications to read its data. So could plug anything to OHM, like a web page for remote monitoring. A feature that could fit .NET 2, keep OHM simple and allow a lot of other stuff to interact (read-only?) with it.