openhardwaremonitor/openhardwaremonitor #1252: Freeze When Display DPI/Resolution changed

Pein1911, 2020-04-18:
My display is 1440p, sometime if i load a game at 1080p fullscreen, then I quit game and  OHM tray icon will be freeze

wingo1, 2020-05-02:
Same problem !

pkypkygit, 2020-05-14:
The same problem here. 

The easiest way to reproduce:
1. set windows DPI scaling to 100%
2. start OpenHWMonitor, minimize it 
3. set windows DPI scaling to 125%
4. OpenHWMonitor is frozen, stops redrawing values in systray

moellerm, 2020-05-24:
This should be fixed in version 0.9.3.

closed 2020-05-24 by moellerm

pkypkygit, 2020-05-25:
Hello.

I have tried it in version 0.9.3 and the problem is still there. Windows TaskManager says that Open Hardware Monitor is "not responding".

Regards.

pkypkygit, 2020-05-25:
Thanks!

Confirmed - I have tried the new version and it works.

Regards.

> 
> 
> This should be fixed in version 0.9.4.