openhardwaremonitor/openhardwaremonitor #1258: Enhancement / Feature Request: "Service Installation, Run As Service"

busysignal, 2020-04-24:
Installation as a service to run and log when machine is running as an appliance without users logged into session.

xsekax, 2020-05-27:
Yes, very useful feature because on windows nothing can collect sensors data in background. With WMI support it will be like lm-sensors in linux.

arkid, 2021-01-15:
Another request for this please, it's my one issue with the program that there isn't built in support for this.  I have a windows PC I use as a home server that sometimes reboots and I don't want it auto logging in.  So often the OHM isn't running due to reboots for software updates etc.

AwfulCharlie, 2021-03-26:
Would also like to see this option

darkbrain-fc, 2021-11-01:
Maybe it's what you're looking for: https://github.com/darkbrain-fc/HardwareSupervisor

busysignal, 2021-11-01:
Where does it log the data? Where are the logs being saved? And can it send the logs directly to a specific server? I like the idea - but I am not seeing the whole picture I believe.

darkbrain-fc, 2021-11-01:
Right now sensors data log are not saved. But I'm plan to implement it in future. Maybe it's not necessary to send whole data to a server, maybe only some interesting one. The idea is interesting but it's not free to rent a server, maybe when a lot of users starts to use HardwareSupervisor can be doable.

busysignal, 2021-11-01:
Sorry, I am looking for something to log the sensors I need locally first in a simple clean CSV file - I have tools that will forward that to my local servers for ingestion into Gravwell... I would like to get that logged locally. Not sending data out to anyone else or the internet. 

Process runs locally, logs locally, if at all possible sends data as a http post in intervals, but mostly just getting it to log and stay locally - - - with a to format the log date-time-rollover-number.

darkbrain-fc, 2021-11-01:
Hmm CSV files are simple but not so easy to elaborate, I thought to use an sqllite instead, maybe with an export utility

busysignal, 2021-11-01:
Not interested - but thanks. Clean flat files so the logging is easy and can be forwarded. all good. thanks for looking out.

darkbrain-fc, 2021-11-01:
Maybe I can think to add that.

darkbrain-fc, 2021-11-01:
BTW now you can create a small script to achieve that. You only need to call 
`get-wmiobject -namespace "root/HardwareSupervisor" -Class Sensor ` to list all sensors, or you can tweak that WMI query to read only sensors you want