openhardwaremonitor/openhardwaremonitor #1261: Enable Hyper-V Windows Features, and Loose CPU Core Clock and Temp Readings

MerlinStWahgwaan, 2020-04-27:
So i began back tracking what I was doing last night and what i was playing with in my laptop that could have caused somthing like this, and I remembered I had been turing on Virtualization and Windows Sandbox so I could use that for testing in a secure environment... anyways....

Thats when I decided to basically undo everything I did last night and see if by chance it solved my issue, and it did.

This is exactly what I did, and I assume it will work for you if you do the same thing; As long as your not using Hyper-V for anything other than Sandbox, which apparently doesnt require any of that running.

Open the Start Menu, [WIN] key, and start typing "Windows Features"
Open "Turn Windows Features On or Off"
Turn off the following (remove the check boxes)

- Guarded Host
- Hyper-V
- Virtual Machine Platform
- Windows Hypervisor Platform

Once I turned those off, and rebooted as required, my OHM worked immediately.

Im not sure which of them is causing OHM to not be able to read the data from the monitor chips, but its obviously somthing to do with Hyper-V being enabled.

Hopfully someone with more knowledge than myself can track down what the issue is and get a proper fix so it can run with hyper-V enabled.

_Originally posted by @MerlinStWahgwaan in https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1242#issuecomment-619679085_

Please see the full, short, thread ref'd above for full details of what the issue what causing.

For myself this moment, I dont need the fully Hyper-V environment, but im sure someone will, so maybe someone should consider looking into why this happens.

MerlinStWahgwaan, 2020-04-27:
Also loss of Power usage readings..... just FYI

saosilvestre, 2020-06-04:
Hi!

Same situation here. In addition to what have already been identified, I have lost as well:

1. GPU temp
2. Battery info (current voltage; design capacity; wear level)

Lukas0610, 2020-06-23:
There's nothing that can be done as Hyper-V (or more exactly the Windows Hypervisor) virtualizes some system resources like the CPU and blocks access to the registers providing those values required to compute the core clock.

Although @saosilvestre the GPU values shouldn't be affected as they are read using the GPU drivers, maybe try updating them.

saosilvestre, 2020-06-24:
Hi Lucas,

GPU drivers: your guess makes sense. I have just checked Intel site and Samsung-custom GPU driver for UHD620 is more than 18 months behind the current one.
I have to confess, I am a bit afraid to install the generic driver – however much it is up to date.