openhardwaremonitor/openhardwaremonitor #1263: SSD Not Recognized

jakesthetester, 2020-05-01:
My main system drive SSD (Samsung 970 EVO Plus 250GB) is not being recognized by any of the last 3 iterations of this software.  All storage drives are still configured exactly as they arrived when they were brand new.  I have another computer on the network that has no problem exposing its SSD (also a Samsung) to OHM.  Will there be a patch or release that can fix this, or am I doing something wrong?  I have included a copy of my .txt file for reference...

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4562126/OpenHardwareMonitor.Report.txt)

Claire83310, 2020-05-05:
Same problem here with my SSD Samsung 970 EVO Plus 500GB
Report & Smart here

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4580457/OpenHardwareMonitor.Report.txt)
[image]
[image]
[image]
[image]

ahmadSum1, 2020-05-22:
Same here with my m.2 nvme SSD, The other SATA 2.5" SSD was recognized properly.
SSD not properly recognized. The "Generic Hard Disk (/hdd/2)" is actually a Gigabyte GP-GSM2NE3128GNTD 128GB,
Firmware: EDFM00.2
[OpenHardwareMonitor.Report_23May2020.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4669646/OpenHardwareMonitor.Report_23May2020.txt)

flexicart, 2020-06-22:
Hello, same issue here, glad I'm not alone! :)

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4814851/OpenHardwareMonitor.Report.txt)
[image]

Sopor, 2020-07-10:
I have the same issue here with my two M.2 drives. OHM will only show Generic hard drive

Speecy seems to detect some of the info but not all as you can see below
```
476GB Samsung SSD 970 PRO 512GB (Unknown (SSD))
465GB NVMe Samsung SSD 970 SCSI Disk Device (Unknown (SSD))
```

jpstotz, 2020-07-31:
It seems like OpenHardwareMonitor does not yet support NVMe SSDs (Windows driver).

NVMe SSDs require different code that regular SATA drives for retrieving detailed info. Example code how to query e.g. the NVMe temperature can be found here: https://stackoverflow.com/a/55241041/150978

farp96, 2020-08-01:
AW that's why mine isn't read. OH well back to HW monitor.

MrCanan, 2020-08-08:
Hi all,
Same for me, OpenHardwareMonitor doesn't support NVMe SSDs, so I got two lines "Generic Hard Disk".
Any news about his support?
Best regards.

jpstotz, 2020-08-08:
There is pull request #984 (opened in 2017) which adds support for NVMe drives that are not using the standard Windows NVMe driver. 

It it seems like there were problems because this PR was never included into OpenHardwareMonitor.

donpin1, 2021-02-19:
Agree, please add NVMe drives and their temperatures. It would be great if you also had a tab that monitored SMART attributes. Nevertheless, this is easily the best program for me that I have come across. Well done.
[image]