openhardwaremonitor/openhardwaremonitor #1265: Feature request: Persist settings across a restart of the application

lbreuss, 2020-05-01:
When I restart the application, the previous configuration is lost, all settings are to their default again.

This affects at least the following settings:
* Menu checkmarks like Minimize to Tray, and Hardware selection

moellerm, 2020-05-05:
Settings should be persisted when the application exits to a file called OpenHardwareMonitor.config. It's a simple xml file with a list of keys/values.

Maybe make sure to manually exit the application once (File -> Exit), in case the system does not provide enough time to write settings to the file when the system is shutting down.

Let me know if the settings are not written at all, or if it is only a problem when the system is shutting down.

lbreuss, 2020-05-06:
OK, todays tests show that the settings file is written. 
Menu checkmark is persisted indeed.
Cannot reproduce, sorry for the hassle.

closed 2020-05-06 by lbreuss