openhardwaremonitor/openhardwaremonitor #1268: Riot's new Vanguard anti cheat blocking this app due to old CPUZ driver

danielcherubini, 2020-05-01:
https://www.reddit.com/r/VALORANT/comments/g9aoap/upcoming_vanguard_changes/foshu6a/

There's a privileged escalation with this driver. 
https://github.com/shareef12/cpuz

Can you please update this driver?

danielcherubini, 2020-05-01:
The issue is in OpenHardwareMonitorLib.sys

[image]

heres a link to the two CVE's

https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2017-15303
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2017-15302

hilmi3x, 2020-05-03:
Is that two CVE's to fix the issue and how can I implement those

moellerm, 2020-05-04:
The OpenHardwareMonitorLib.sys is actually the original WinRing0 driver. The source code can be found here: https://github.com/openhardwaremonitor/openhardwaremonitor/tree/master/External/WinRing0

To change the driver you do not only need to build the source code, but you need to sign it with a code signing certificate issued by a trusted root authority. That's at least what I remember, Microsoft seems to change a certain things for 2021. And I do not have such a certificate.

moellerm, 2020-05-06:
Besides the difficulties to rebuild and sign the driver, there is also the problem of what to fix exactly. 

So far I have only seen that Riot Vanguard does not like the WinRing0 driver that the Open Hardware Monitor uses to access the hardware and blocks it. I haven’t seen a specification of what Riot Vanguard does not like about the driver. Obviously this would be needed before trying to start implementing a fix.

R3al1ty, 2020-05-07:
I researched this a bit and I think it is in connection to the following report:

https://safebreach.com/Post/HP-Touchpoint-Analytics-DLL-Search-Order-Hijacking-Potential-Abuses-CVE-2019-6333

https://nvd.nist.gov/vuln/detail/CVE-2019-6333

Interestingly, they mention this project as being used by the HP touchpoint software and even reference this project's github page.

This second report also mentions this project and the WinRing0 driver specifically as allowing an attacker to gain Kernel Mode privileges from a user mode application:

https://eclypsium.com/2019/11/12/mother-of-all-drivers/
https://github.com/eclypsium/Screwed-Drivers

Cheat developers that Vanguard is trying to protect against seem to be using exploits in drivers such as winring0 to gain privileges higher than most anti-cheat software (which often still runs in usermode) to allow full read-write access to the entire device memory while remaining undetected.
Vanguard is blocking all such vulnerable drivers, including winring0 and an older cpu-z driver used in applications from vendors such as corsair.

Riot's blog posts:
https://na.leagueoflegends.com/en-us/news/dev/dev-null-anti-cheat-kernel-driver/
https://support-valorant.riotgames.com/hc/en-us/articles/360046160933-What-is-Vanguard-
https://www.riotgames.com/en/news/a-message-about-vanguard-from-our-security-privacy-teams

danielcherubini, 2020-05-07:
I'm amazed that the security researchers knew the driver is from this project, but instead just focused on reporting it to HP instead of to here also... I guess it's the potential bounty they were after.

Thanks @R3al1ty for finding the CVE, I couldn't find it when searching for it, instead I assumed (incorrectly) that this project was also using the CPU-Z driver. I'm sorry for the confusion.

R3al1ty, 2020-05-07:
After skimming the code of OpenHardwareMonitor, I don't think there is a simple "fix" which would make riot happy.

The main problem is that winring0 (https://github.com/QCute/WinRing0) pretty much allows arbitrary read/write access to all devices and system memory with no limitations. This is exactly what riot wants to prevent with their anti-cheat - so as long as a driver allows arbitrary memory r/w from user-space, riot will block that driver.

As I understand it, OpenHardwareMonitor needs this read/write access to read and write pretty much arbitrary memory for sensor values and fan speeds of a huge variety of different devices.
The only solution would be a new driver that exposes a very limited API in which all the r/w logic of sensor value extraction is on the driver side.
But as @moellerm already mentioned, building a new driver is expensive and highly complicated because the driver would need to be signed by Microsoft...

moellerm, 2020-05-07:
I haven't seen a notification about the problems mentioned in https://nvd.nist.gov/vuln/detail/CVE-2019-6333

Beside this, there is also this issue: https://nvd.nist.gov/vuln/detail/CVE-2019-7240 which does not reference the Open Hardware Monitor, but I think still applies. The Open Hardware Monitor uses the very same driver (WinRing0).

Regarding CVE-2019-6333: From what I understand, there are a few details/claims wrong. The WinRing0 driver does not allow writing to physical memory at all, the corresponding IOCTL is disabled in the driver. The driver allows reading arbitrary memory, but only in the address range 0x000C0000 - 0x000FFFFF (256 KB). 

The potentially loading arbitrary dlls into the OHM process (that is usually run with admin rights) is only one part of the problem. For the problem to occur, an admin has to include a directory (with write access to non-admins) into the PATH variable. An attacker can then place a dll (with the right name) in that folder, and OHM risks to load it.

But there are other situations that are more critical, when the OHM binaries are stored in a directory with write access to non-admins and (automatically) started at Windows start-up with admin rights from there.

R3al1ty, 2020-05-07:
Ah, right, that sounds a lot different than I made it out to be, sorry. The readme found in https://github.com/QCute/WinRing0 sounds a lot more dangerous.

moellerm, 2020-05-07:
Look at the source code

For memory read access: https://github.com/openhardwaremonitor/openhardwaremonitor/blob/master/External/WinRing0/OpenLibSys.c#L605

For memory write access: https://github.com/openhardwaremonitor/openhardwaremonitor/blob/master/External/WinRing0/OpenLibSys.c#L652

As far as I know the WinRing0 driver used in OHM is built without the _PHYSICAL_MEMORY_SUPPORT. The file has been originally compiled for the OpenLibSys project.

moellerm, 2020-05-07:
The problem CVE-2019-7240 described here 
https://github.com/fireeye/Vulnerability-Disclosures/blob/master/FEYE-2019-0004/FEYE-2019-0004.md
is more critical and still present in WinRing0. This could be fixed easily with a new driver. But the general problem of a very low level API with direct MSR, IO port and PCI access remains. I think it is not really possible to make this type of API completely secure while maintaing the current functionality of the OHM. 

The best thing would be to move all code that requires direct hardware access into the driver, and maintain only a very high level API. But then the driver would have to be rebuilt and signed for every change, something that seems not very practical.

moellerm, 2020-05-07:
But overall, as long as drivers like the current WinRing0 are not blocked on an operating system level, there is very little to be gained in terms of general security by replacing WinRing0 with a new driver. Any process with admin rights could still load the original WinRing0 driver.

And regarding Riot Vanguard: I have some doubts about their general approach at trying to prevent cheating by blacklisting certain drivers. And while we don't have a clear specification of what would prevent any future driver from getting blacklisted, I don't see much motivation to get work started on this. I guess for now one has to choose to use either OHM (with low level hardware access), or Riot Vanguard.

closed 2020-05-07 by moellerm

moellerm, 2020-05-07:
Regarding the WinRing0 driver issues, check also the issue #46.

masterjunior24, 2020-05-08:
Riot pushed an update to vanguard that doesn't block the driver anymore and many other drivers
https://www.reddit.com/r/VALORANT/comments/gfesag/when_this_post_is_1_hours_old_riot_will_release_a/