openhardwaremonitor/openhardwaremonitor #1269: Bug CPU

Brice3666, 2020-05-02:
[image]
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4566905/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2020-05-02:
Ring0

Status: Installing driver "C:\Users\PCBrice\Desktop\OpenHardwareMonitor\OpenHardwareMonitorLib.sys" failed and file exists
First Exception: Service already exists
Second Exception: StartService returned the error: Le pilote nâ€™a pas pu Ãªtre chargÃ©, car son appel de lâ€™initialisation a Ã©chouÃ©. (Exception de HRESULT : 0x80070287)

Brice3666, 2020-05-02:
how i made it work

DarthDestroyer, 2020-05-02:
Check if anything is blocking the Service!
Anti Virus? Any kind of Anticheat...

Fhosho, 2020-05-05:
I have the exact same problem! It works when i System restore my computer just one day back, but when i go to sleep and wake up the next morning the issue is right there again! Solutions ?

moellerm, 2020-05-05:
Thx for the feedback. This seems to be a duplicate of #1268

closed 2020-05-05 by moellerm