openhardwaremonitor/openhardwaremonitor #1270: CPU Temps/Clocks stopped working

danielquijada, 2020-05-02:
I mainly use this program to keep track of the temperatures, so I'm sure it worked a few days ago, and I haven't changed anything in my PC.

Here's the report: 
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4567693/OpenHardwareMonitor.Report.txt)

I hope this is solved soon, because I really like this program =(

If anyone has any fix that have worked for them, please let me know and I'll try them.

DarthDestroyer, 2020-05-02:
Ring0

Status: Installing driver "D:\Descargas\Chrome\openhardwaremonitor-v0.9.2\OpenHardwareMonitor\OpenHardwareMonitorLib.sys" failed and file exists
First Exception: Service already exists
Second Exception: StartService returned the error: El controlador no se descargÃ³. Error en la llamada de inicializaciÃ³n. (ExcepciÃ³n de HRESULT: 0x80070287)

Check if anything is blocking the Service! 
Anti Virus? Any kind of Anticheat...

DarthDestroyer, 2020-05-02:
Duplicate of #1269

danielquijada, 2020-05-02:
That was it! Riot Vanguard (Riot's anticheat system) apparently was blocking it. I uninstalled Vanguard and now it works..

I would love not to have to choose between Vanguard and OHM, but this works for me in the meantime. Thanks!

PS: u're right, this is a duplicate, my bad, I'm closing it

closed 2020-05-02 by danielquijada