openhardwaremonitor/openhardwaremonitor #1271: Fan Controls

alkatena, 2020-05-04:
Hi!

First of all, I'd like to thank you all for this great project you've created!

I'm developing an application using OHM as a backend layer for a mobile app. Everything works perfectly!

However, there's one thing I'm not able to figure out. I want to show the CPU Fan speed as percentage.  I believe this information should be in the `Controls` part, as can be seen here:

[image]

Controls' information is not showing as you can see in the above image. Fan speed shows correctly. I could calculate % by knowing the max RPM for the FAN. But I don't have that information. I could say to the fan "full power" and wait some seconds to read the RPM at this state and then I'd have the max RPM. But I don't know how to do this and if it's the best way to do it. (I imagine the motherboard does this when system boots and then stores this value somewhere).

However, when I use the Gigabyte System Information View program, it does show that information:

[image]

You can even notice Gigabyte System Information View seems to detect the same 3 fans that OHM does, which are all set to 0 (as I only have a CPU fan connected) and two more:

[image]

So I think there must be an issue or something missing in OHM in order to correctly detect CPU Fan Control (because the speed is detected ok).

I need help for trying to make this work!

Many thanks in advance

PD: Here's the link to my report (nothing seems to be wrong): [link](https://pastebin.com/Nncbc7Qx)

moellerm, 2020-05-17:
It's possible that Gigabyte System Information View retrieves the max RPM from the BIOS. But this is most likely undocumented. 

In the Open Hardware Monitor, the (commanded) fan speed percentage is shown only when the fans are controlled by software. For the "default" mode (fan controlled by the IT8628E itself), the commanded speed percentage is not available. Because of this, nothing is displayed.

Generally, the max RPM information has to be measured by running the fans at full power for some time. The way PC fans are built and connected to the mainboard, this is the only way to obtain this information.

closed 2020-05-17 by moellerm

mb-asapp, 2020-05-21:
Yeah, I end up doing so! Thanks