openhardwaremonitor/openhardwaremonitor #1272: Merge with fork (LibreHardwareMonitor)

xcasxcursex, 2020-05-05:
This is a good project but [was essentially dead for several years](https://github.com/openhardwaremonitor/openhardwaremonitor/graphs/code-frequency). In that time, a fork was made in order to maintain constant updates: https://github.com/LibreHardwareMonitor/

Since this project seems to have woken from the dead, it has been suggested that the two projects merge. [Unfortunately, the author of that project has said that there is no interest in doing so, from this side.](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/240#issuecomment-624049725)

That's very unfortunate. Perhaps you would reconsider. Is there some reason why the two projects should not be merged?

informatorius, 2020-05-05:
Or if merge is not wanted then just take missing code from LibreHardwareMonitor and integrate here

moellerm, 2020-05-05:
From my perspective there are several issues with the LHM code base (which I have also communicated to the LHM maintainer):

- Renaming of files and folders in LHM have broken git history. This makes it very difficult to find original commits that introduced certain code parts (and potential reasons for doing so).

- Code style changes in LHM: The code style used in OHM (K&R braces, 80 characters line length, 2 spaces tab) had been chosen intentionally and I would still consider this a good choice at this point.

- Some functionality has been removed in LHM: Crash and report submission forms for example. Some of the "code style" changes introduced regressions, and the corresponding commits are far to huge to review in all details.

- All the copyright notices in the source code files had been removed in LHM. This is not allowed under MPL 2.0 license (see section 3.4 of the license for details). The copyright notices have been "restored" to some modified form now, but in my opinion this is still in slight conflict with the MPL.

 - The source files should use the license notice from Exhibit A of the MPL 2.0 license, see also section 1.4 of the license. This is not the case in the current LHM code base.

- At the moment the OHM code base still supports .NET 2.0 systems. 

For these reasons I do not want to drop the current OHM code base and replace it with the LHM code base. And the LHM project does not want to continue work based on the OHM code base. 

I will accept individual pull requests for individual features and improvements into OHM, with the exception of whole project code style changes and (currently) raising the .NET framework requirement.

I will continue development of missing features into OHM as time permits.

DoubleOhmSeven, 2020-05-06:
My 2 cents.

>     * Renaming of files and folders in LHM have broken git history. This makes it very difficult to find original commits that introduced certain code parts (and potential reasons for doing so).

This is a non-issue, you can create a new branch and it's not like you cannot search in it.

>     * Code style changes in LHM: The code style used in OHM (K&R braces, 80 characters line length, 2 spaces tab) had been chosen intentionally and I would still consider this a good choice at this point.
 
Probably nobody else finds this a good choice. This is far from the C# default and this is the only project I know of that uses this _unreadable_ code style.

>     * Some functionality has been removed in LHM: Crash and report submission forms for example. Some of the "code style" changes introduced regressions, and the corresponding commits are far to huge to review in all details.

Pretty sure that can be added back quite easily. There are no regressions I've encountered, do you have examples of this?

> * All the copyright notices in the source code files had been removed in LHM. This is not allowed under MPL 2.0 license (see section 3.4 of the license for details). The copyright notices have been [...]

It's totally fine like this in MPL. As long as MPL notice is there with reference to original project you may alter to correct it. Although libre is probably a larger work which means that all of this would be void anyway.
 
>     * At the moment the OHM code base still supports .NET 2.0 systems.

.NET 2.0 hasn't been supported **since 2011** and has security issues and other faults. There's a reason people want to update to new .NET versions.

> I will continue development of missing features into OHM as time permits.

This lines out the problem we've seen in the past: a couple of years of no activity. That's why you want multiple maintainers to a project, so everything can continue.

moellerm, 2020-05-06:
Maybe the I'm not familiar enough with git, but this is my issue with the git history. For example, checking for revisions of the F718XX class produces the following in OHM (one click):

[image]

while in LHM I can only see up to the code style change (and file renaming):

[image]

which excludes any history before September 2019.

I won't relaunch a detailed code style discussion here, you can find plenty of detailed discussions elsewhere. The use of K&R braces might be rarer for C# projects, but very common for Java. With the Allman style it's just the opposite. With the two languages Java and C# being quite similar, I don't see how one of the two styles can be considered objectively more readable than the other.

An example of a regression introduced with the "code style" commit has been fixed here: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/61454e1ae2f2ce18550bbb337719e39a3058884f

Regarding the MPL: I'm not sure how you are allowed to alter or remove copyright notices, given section 3.4 of the MPL 2.0. The alterations have not been done to "to remedy known factual inaccuracies".

"3.4. Notices
You may not remove or alter the substance of any license notices (including copyright notices, patent notices, disclaimers of warranty, or limitations of liability) contained within the Source Code Form of the Covered Software, except that You may alter any license notices to the extent required to remedy known factual inaccuracies."

Where in the MPL does it void the requirements from section 3.4 for "larger works"? Anyway, this is just a detail and is not really what is blocking a merge.

Support for .NET 2.0 might get removed from OHM if another feature is blocked by this. Keeping the .NET version requirements low gives a better chance that the OHM will just work on a newly installed system (for a quick check for example) without having to install .NET frameworks first.

I agree that we need more maintainers on the project.

DoubleOhmSeven, 2020-05-06:
> Maybe the I'm not familiar enough with git, but this is my issue with the git history. For example, checking for revisions of the F718XX class produces the following in OHM (one click):
>
> [image] [...]

This is because the file is no longer at the same location, you can still browse the history with git, you just need to use earlier commit.
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/tree/d92d88d905087dc797bf6625ad85b54b776c4210/
If you want it easier, you just create another branch with new code, then you can easily switch to old branch and search there.

> I won't relaunch a detailed code style discussion here, you can find plenty of detailed discussions elsewhere. The use of K&R braces might be rarer for C# projects, but very common for Java. With [...]

Possibly, but this is C# code and not Java. If you want to attract more maintainers then it should be easy to get into and the current code style ain't very readable and Visual Studio complains about code style everywhere.
> 
> An example of a regression introduced with the "code style" commit has been fixed here: [LibreHardwareMonitor/LibreHardwareMonitor@61454e1](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/61454e1ae2f2ce18550bbb337719e39a3058884f)

Which is fixed there so that's hardly a reason for it.

> Support for .NET 2.0 might get removed from OHM if another feature is blocked by this. Keeping the .NET version requirements low gives a better chance that the OHM will just work on a newly installed system (for a quick check for example) without having to install .NET frameworks first.

I get that, but how much of the user base doesn't have a newer one installed? Pretty sure everyone has at least 452 installed by now (which isn't even supported anymore by Microsoft).

> I agree that we need more maintainers on the project.

Yes. There's several active maintainers at libre 👍

xcasxcursex, 2020-05-06:
While it may take some effort, I don't see any of these issues preventing a merge. Code style and licensing are of no consequence to the user. Removed features can be re-added. .NET prerequisites can be installed with the application. Etc etc....there are ways around all this. The real issue here is a lack of desire to overcome the issues and make the merge.

I don't see that having these two projects coexisting is beneficial for anyone concerned. It's a real shame.

closed 2020-05-06 by moellerm

MoshiMoshi0, 2020-05-09:
I know this issue is closed but I'll add my 2 cents.

> Code style changes in LHM: The code style used in OHM (K&R braces, 80 characters line length, 2 spaces tab) had been chosen intentionally and I would still consider this a good choice at this point.

My go to languages before C# were C/C++ and java and while I think C#'s code style wastes a lot of vertical space I think open source projects should use default conventions for used languages and tools. In this case the default code style should be the one suggested by microsoft. You can/should of course tweak the style per line/file where it can improve readability.

> At the moment the OHM code base still supports .NET 2.0 systems.

This is not something I would consider a good thing. If someone has a need to run OHM on an old server that cant be updated, he wont need the features of newer OHM anyway so he can just use the last version that supports his desired .net framework. I would just drop old framework versions after some time and move them to a separate branch that only receives critical reported fixes. There are a lot of features that can improve readability or sometimes performance in newer .net of C# versions.

C# is transitioning to .net core and soon to .net 5 so there is no point supporting very old .net frameworks. You said that you want to support fresh systems but you can just create a self-contained build with .net core libs packed. I hope someday OHM and/or LHM can be fully ported to .net 5 and that they will fully work on multiple platforms.