openhardwaremonitor/openhardwaremonitor #1274: Max Fan Speed showing incorrect readings

TranquilHegemony, 2020-05-11:
I have got an issue with OpenHardware Monitor showing a random reading for two of the the three fans that it is reading. I suspect that the issue is down to the fans that are providing the readings being joined by a splitter cable and this is causing the issue. The one fan that I have that is a single fan to the MOBO header doesn't show the issue. All the fans in questions are NF-A14 PWM - all were ordered at the same time and all cables are Noctua as supplied with the fans.

Just to add that when it OHM starts they show the correct Max figures but after a period of time they then start to show as shown in the attached. 

[image]

Thanks 

TranquilHegemony

moellerm, 2020-05-14:
Please attach report file created with File -> Save report...

Do you run other monitoring tools in parallel when this happens?

Splitter cable should normally not be an issue, if it is a standard one. The splitter cable only connects the tachometer signal from one of the two fans to the mainboard. So you can monitor the RPMs of one fan only, for the other the mainboard is blind.

TranquilHegemony, 2020-05-14:
Please see attached.

Error occurred within a couple of minutes approximately of launching
openhardwaremonitor.

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.9.2.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.18363.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- MSI Z390-A PRO (MS-7B98) (/mainboard)
|  |
|  +- Nuvoton NCT6797D (/lpc/nct6797d)
|  |  +- CPU VCore      :    0.688     0.68    1.256 (/lpc/nct6797d/voltage/0)
|  |  +- Voltage #2     :    1.016    1.008    1.016 (/lpc/nct6797d/voltage/1)
|  |  +- AVCC           :    3.344    3.344    3.344 (/lpc/nct6797d/voltage/2)
|  |  +- 3VCC           :     3.36    3.344     3.36 (/lpc/nct6797d/voltage/3)
|  |  +- Voltage #5     :    1.016    1.016    1.016 (/lpc/nct6797d/voltage/4)
|  |  +- Voltage #6     :    0.136    0.136    0.144 (/lpc/nct6797d/voltage/5)
|  |  +- Voltage #7     :     0.96     0.96    0.968 (/lpc/nct6797d/voltage/6)
|  |  +- 3VSB           :    3.344    3.344    3.344 (/lpc/nct6797d/voltage/7)
|  |  +- VTT            :    1.064    1.064    1.064 (/lpc/nct6797d/voltage/9)
|  |  +- Voltage #11    :    1.064    1.064    1.064 (/lpc/nct6797d/voltage/10)
|  |  +- Voltage #12    :    1.064    1.064    1.064 (/lpc/nct6797d/voltage/11)
|  |  +- Voltage #13    :    1.048    1.048    1.048 (/lpc/nct6797d/voltage/12)
|  |  +- Voltage #14    :      0.6      0.6      0.6 (/lpc/nct6797d/voltage/13)
|  |  +- Voltage #15    :    1.536    1.528    1.536 (/lpc/nct6797d/voltage/14)
|  |  +- CPU Core       :       23       23       34 (/lpc/nct6797d/temperature/0)
|  |  +- Temperature #1 :     24.5       24       25 (/lpc/nct6797d/temperature/1)
|  |  +- Temperature #2 :       31       31       32 (/lpc/nct6797d/temperature/2)
|  |  +- Temperature #3 :       28       28       28 (/lpc/nct6797d/temperature/3)
|  |  +- Temperature #4 :       22       22       22 (/lpc/nct6797d/temperature/4)
|  |  +- Temperature #5 :       22       22       23 (/lpc/nct6797d/temperature/5)
|  |  +- Temperature #6 :       -2       -2       -2 (/lpc/nct6797d/temperature/6)
|  |  +- AIO Pump       :     1928     1282     1928 (/lpc/nct6797d/fan/0)
|  |  +- Exhaust Fan    :      437      430      447 (/lpc/nct6797d/fan/4)
|  |  +- Intake Fans 1  :      348      344      375 (/lpc/nct6797d/fan/5)
|  |  +- Intake Fans 2  :      345      345    19640 (/lpc/nct6797d/fan/6)
|  |  +- Fan Control #1 :      100      100      100 (/lpc/nct6797d/control/0)
|  |  +- Fan Control #2 :  12.9412  12.9412  12.9412 (/lpc/nct6797d/control/1)
|  |  +- Fan Control #3 :  57.6471  57.6471  57.6471 (/lpc/nct6797d/control/2)
|  |  +- Fan Control #4 :  57.6471  57.6471  57.6471 (/lpc/nct6797d/control/3)
|  |  +- Exhaust Fan    :  24.7059  24.7059  26.6667 (/lpc/nct6797d/control/4)
|  |  +- Intake Fans 1  :       20       20  23.1373 (/lpc/nct6797d/control/5)
|  |  +- Intake Fans 2  :       20       20  23.1373 (/lpc/nct6797d/control/6)
|
+- Intel Core i7-9700K (/intelcpu/0)
|  +- Bus Speed      :      100      100      100 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  800.001      800  4800.01 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :  800.001      800  4800.01 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :  800.001      800  4800.01 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  800.001      800  4800.01 (/intelcpu/0/clock/4)
|  +- CPU Core #5    :  800.001      800  4800.01 (/intelcpu/0/clock/5)
|  +- CPU Core #6    :  800.001      800  4800.01 (/intelcpu/0/clock/6)
|  +- CPU Core #7    :  800.001      800  4800.01 (/intelcpu/0/clock/7)
|  +- CPU Core #8    :  800.001      800  4800.01 (/intelcpu/0/clock/8)
|  +- CPU Core #1    :       22       22       33 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       23       22       34 (/intelcpu/0/temperature/1)
[... 1109 lines cut ...]
 B6 Erase Fail Count (Total)           000000000000 100   100   10    0       
 B7 Runtime Bad Block Total            000000000000 100   100   10    0       
 BB Uncorrectable Error Count          000000000000 100   100   0     0       
 BE Temperature                        180000000000 60    76    0     24      
 C3 ECC Rate                           000000000000 200   200   0     -       
 C7 CRC Error Count                    010000000000 99    99    0     1       
 EB Power Recovery Count               1F0000000000 99    99    0     -       
 F1 Total LBAs Written                 302FB1820000 99    99    0     1045.537

Logical drive name: G:\
Format: NTFS
Total size: 500105736192
Total free space: 499986239488

--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Generic Hard Disk
Firmware version: Unknown

Logical drive name: C:\
Format: NTFS
Total size: 249445740544
Total free space: 191696146432

TranquilHegemony, 2020-05-14:
Missed off the reply to the other monitoring software question. Only
running openhardwaremonitor.

Thanks

Lee

moellerm, 2020-05-14:
Thx for the report. Unfortunately I do not really have an idea what is causing these wrong readings. 

The values from the registers:

```
 RPM from Counter (13-bit)
 Fan        1       2       3       4       5       6       7
         1928       0       0       0     442     351   26470 
 Registers 		 
 04B0   15 1C   FF 1F   FF 1F   FF 1F   5F 0E   78 01   00 33 

 
 RPM from RPM Reading (16-bit)
 Fan        1       2       3       4       5       6       7	
         1928       0       0       0     442     351   31237 
 Registers		 
 04C0   07 88   00 00   00 00   00 00   01 BA   01 5F   7A 05 
```
Currently only the direct RPM reading is used (and displayed). Maybe one could try to validate values by comparing with values computed from the counter value to filter "wrong" values. But this seems more like a fix to hide the problem.

moellerm, 2020-05-15:
The counter register for the last fan "Intake Fans 2" (/lpc/nct6797d/fan/6) has invalid values, the max value for the low-bits register is 0x1F (5-bits). Does the "Intake Fans 2" ever show reasonable RPM values? It might be that these register are used for something else on the NCT6797D.

But this still does not explain the 19640 value on "Intake Fans 1". And in the report you sent me, I can't see the values in the current register values. It has only been present at some point in the past (max values).

TranquilHegemony, 2020-05-16:
It will always be reasonable at start-up for Intake Fans 1 and Intake Fans
2. Then at some point (I will say 5 minutes as a very rough guess but I
haven't timed it) the Max Speed for one or both will then show the high
figure. The odd thing is that it will usually only be on one of these and
the other will show a reasonable reading for longer and then at some point
that will also start to show the high figure. This only happens on these
two fans, all others are fine and as previously mentioned these are the
only ones running the Noctua splitter cables hence my suspicion that these
are somehow causing the issue but no idea how - as you have already said
the reading is only taken from one of the two with the other not providing
any input re: fan RPM.

Should add that it isn't a massive concern to me so don't feel as if you
have to devote a huge amount of time to this - it is more of an oddity that
I had noticed so i thought I would flag it with you. I know that the
reading is incorrect and can make allowances for it and it is not one I
would consider important should i ever have to investigate cooling issues.

Will also look to see if there are any BIOS upgrades available for the
motherboard - MSI Z390-A Pro - that can be applied. If there is then i will
rerun the tests and update.

Thanks

Lee

moellerm, 2020-05-16:
Nvm my comment above about invalid values for the "Intake Fans 2" (/lpc/nct6797d/fan/6), I looked at the wrong register. 

At the origin of the problem seems to be a fan tachometer signal that stops the 13-bit counter after just two counts. This would correspond to an RPM value of 675'000. The NCT6797D then tries to put this value into the 16-bit RPM register, but since it is too high (largest value it could hold is 65'535 RPM), it then truncates the value to 16-bit, which produces 19'640 RPM.

I have changed the way the fan RPMs are read for this series of Nuvoton chips by reading directly the 13-bit counter register instead of the 16-bit calculated RPM (and potentially truncated) value.

You can try with this new version:
https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.3.2-alpha.zip

Btw. I'm not sure why the fan tachometer signal counter stops sometimes already after two counts. It might be an issue of a bad signal/contact/interference. But this is really hard to tell, and I'm not aware of a method to really fix the source of this issue. In the new version linked above I discard values with unrealistic high RPMs (anything above 64'286 RPMs), so bad readings like that should not show up in max/min values.

Please let me know how this new version behaves.

TranquilHegemony, 2020-05-24:
Sorry for the delay replying but have been rather busy this week.

The new version has resolved the issue with the invalid values - those have
ben stable since running the new version.

Seems to be hit and miss with the pump speed though - it didn't show at all
when i launched the the new version but appeared sometime after OK as was
showing values around 1900RPM. Have since done a reset and initially it
didn't show but it is now showing but 0RPM across the board.

Screen shot attached and a new Report included as well.

Thanks

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.9.2.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.18363.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- MSI Z390-A PRO (MS-7B98) (/mainboard)
|  |
|  +- Nuvoton NCT6797D (/lpc/nct6797d)
|  |  +- CPU VCore      :    0.688     0.68    1.256 (/lpc/nct6797d/voltage/0)
|  |  +- Voltage #2     :    1.016    1.008    1.016 (/lpc/nct6797d/voltage/1)
|  |  +- AVCC           :    3.344    3.344    3.344 (/lpc/nct6797d/voltage/2)
|  |  +- 3VCC           :     3.36    3.344     3.36 (/lpc/nct6797d/voltage/3)
|  |  +- Voltage #5     :    1.016    1.016    1.016 (/lpc/nct6797d/voltage/4)
|  |  +- Voltage #6     :    0.136    0.136    0.144 (/lpc/nct6797d/voltage/5)
|  |  +- Voltage #7     :     0.96     0.96    0.968 (/lpc/nct6797d/voltage/6)
|  |  +- 3VSB           :    3.344    3.344    3.344 (/lpc/nct6797d/voltage/7)
|  |  +- VTT            :    1.064    1.064    1.064 (/lpc/nct6797d/voltage/9)
|  |  +- Voltage #11    :    1.064    1.064    1.064 (/lpc/nct6797d/voltage/10)
|  |  +- Voltage #12    :    1.064    1.064    1.064 (/lpc/nct6797d/voltage/11)
|  |  +- Voltage #13    :    1.048    1.048    1.048 (/lpc/nct6797d/voltage/12)
|  |  +- Voltage #14    :      0.6      0.6      0.6 (/lpc/nct6797d/voltage/13)
|  |  +- Voltage #15    :    1.536    1.528    1.536 (/lpc/nct6797d/voltage/14)
|  |  +- CPU Core       :       23       23       34 (/lpc/nct6797d/temperature/0)
|  |  +- Temperature #1 :     24.5       24       25 (/lpc/nct6797d/temperature/1)
|  |  +- Temperature #2 :       31       31       32 (/lpc/nct6797d/temperature/2)
|  |  +- Temperature #3 :       28       28       28 (/lpc/nct6797d/temperature/3)
|  |  +- Temperature #4 :       22       22       22 (/lpc/nct6797d/temperature/4)
|  |  +- Temperature #5 :       22       22       23 (/lpc/nct6797d/temperature/5)
|  |  +- Temperature #6 :       -2       -2       -2 (/lpc/nct6797d/temperature/6)
|  |  +- AIO Pump       :     1928     1282     1928 (/lpc/nct6797d/fan/0)
|  |  +- Exhaust Fan    :      437      430      447 (/lpc/nct6797d/fan/4)
|  |  +- Intake Fans 1  :      348      344      375 (/lpc/nct6797d/fan/5)
|  |  +- Intake Fans 2  :      345      345    19640 (/lpc/nct6797d/fan/6)
|  |  +- Fan Control #1 :      100      100      100 (/lpc/nct6797d/control/0)
|  |  +- Fan Control #2 :  12.9412  12.9412  12.9412 (/lpc/nct6797d/control/1)
|  |  +- Fan Control #3 :  57.6471  57.6471  57.6471 (/lpc/nct6797d/control/2)
|  |  +- Fan Control #4 :  57.6471  57.6471  57.6471 (/lpc/nct6797d/control/3)
|  |  +- Exhaust Fan    :  24.7059  24.7059  26.6667 (/lpc/nct6797d/control/4)
|  |  +- Intake Fans 1  :       20       20  23.1373 (/lpc/nct6797d/control/5)
|  |  +- Intake Fans 2  :       20       20  23.1373 (/lpc/nct6797d/control/6)
|
+- Intel Core i7-9700K (/intelcpu/0)
|  +- Bus Speed      :      100      100      100 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  800.001      800  4800.01 (/intelcpu/0/clock/1)
[... 1118 lines cut ...]
 B6 Erase Fail Count (Total)           000000000000 100   100   10    0       
 B7 Runtime Bad Block Total            000000000000 100   100   10    0       
 BB Uncorrectable Error Count          000000000000 100   100   0     0       
 BE Temperature                        180000000000 60    76    0     24      
 C3 ECC Rate                           000000000000 200   200   0     -       
 C7 CRC Error Count                    010000000000 99    99    0     1       
 EB Power Recovery Count               1F0000000000 99    99    0     -       
 F1 Total LBAs Written                 302FB1820000 99    99    0     1045.537

Logical drive name: G:\
Format: NTFS
Total size: 500105736192
Total free space: 499986239488

--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Generic Hard Disk
Firmware version: Unknown

Logical drive name: C:\
Format: NTFS
Total size: 249445740544
Total free space: 191696146432

moellerm, 2020-05-24:
The fans are not displayed initially when they read as zero RPM.

I don't see your screenshot. 

I have just released a new version (0.9.4), please do any further testing with this new version.

I am closing this, as the issue with the wrong max speed values seems to be resolved. If you notice any other problems, please open a new issue.

closed 2020-05-24 by moellerm