openhardwaremonitor/openhardwaremonitor #1275: Asrock X470 Gaming K4 and Ryzen 3700X wrong CPU temperatures..

tonyshark78, 2020-05-14:
Open Hardware Monitor shows different temperatures from Ryzen Master software on my AMD Ryzen 3700X. In the image the CPU is under load and Ryzen Master is showing 74C, Open Hardware Monitor 0.9.3  51C for CPU and 52.5C for what?
Then there were also bogus temperatures sensors that I've hidden (-20C...+200C...)

[image]

moellerm, 2020-05-14:
Please attach a report file. Create it with "File -> Save Report ..."

"Temperatures" from the mainboard super I/O (NCT6779D in this case) are usually completely undocumented and difficult to interpret.

Ryzen Master shows (some variant) of CPU core temperatures. If you want to compare, please compare with temperatures listed directly under the CPU in OHM.

tonyshark78, 2020-05-14:
my report file 

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4630230/OpenHardwareMonitor.Report.txt)