openhardwaremonitor/openhardwaremonitor #1276: Fan speed control loses track while sleep

mhk76, 2020-05-14:
After my computer gets back from sleep, manual fan controls have lost their control over the speed. Menu still shows that the manual control is on, but it really isn't. I have to change the value and it works again.

Basically this needs tracking of Windows' power mode event (SystemEvents.PowerModeChanged) and reset the manual control values when power gets back.

moellerm, 2020-05-14:
Can you please attach a report file? Create one with File -> Save Report...

Otherwise it is difficult to know which hardware is affected by this.

Ale-Rock, 2020-05-16:
Hi, I join the conversation that I'm having the same problem. how can it be solved?
I attach the report file.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4638664/OpenHardwareMonitor.Report.txt)

moellerm, 2020-05-17:
I have tried to fix this in a new alpha version:

https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.3.3-alpha.zip

This version resets the complete OHM hardware tree when the system resumes from sleep.

Please let me know if this works as intended.

mhk76, 2020-05-17:
Works for me. Great work.

Ale-Rock, 2020-05-18:
It works for me too. Thank you very much.

moellerm, 2020-05-18:
Thx for the feedback.

closed 2020-05-18 by moellerm