openhardwaremonitor/openhardwaremonitor #1277: False CCDs created, reading 204.9c - Ryzen 9 3900x

Alpha-087, 2020-05-17:
I'm going to go ahead and assume this is a bug of some sort with the software?

I noticed this started happening after updating my BIOS and chipset drivers.

My mobo is ASUS TUF GAMING X570-PLUS (WI-FI) - BIOS vers 1407
And my CPU is a Ryzen 9 3900x 
 
[image]

Alpha-087, 2020-05-17:
Here is my attached report as well.  
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4640761/OpenHardwareMonitor.Report.txt)

moellerm, 2020-05-17:
Looks to me like a collision with the hardware access. If another software tries to read the CPU package temperature at the same time, this will read 0x0FEF where it expects the values for the CCD temperatures. This value has the "valid" flag (0x800) set, and the temperature results in 204.875 °C ((0x0FEF & 0x7FF) * 0.125 - 49). The CCD sensors remain visible if one "valid" value has been seen there once (even if it currently does not provide valid values).

Does this happen as well if you do not run CoreTemp in parallel?

I can see if there are any accepted ways to synchronize with other software for this hardware access.

Alpha-087, 2020-05-17:
Yes, this happened when I wasn't running Core Temp, though it had only generated one additional CCD that I noticed.  I only installed Core Temp after I noticed this happen, to see if it reported any similar issue.  Beyond that, I didn't have any other temp monitoring software running.

moellerm, 2020-05-17:
I have added support for the mutex "Global\\Access_PCI" to interlock access to the PCI bus. This should avoid conflict with other tools that support this same mutex. I think Core Temp supports this mutex as well. You can test with this new version:

https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.3.4-alpha.zip

Please let me know the results.