openhardwaremonitor/openhardwaremonitor #1281: CPU temperature not showing for i7-10875h

nnchiew, 2020-05-23:
There is no showing for the CPU (ex: clocks, temperature etc. except load)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4671509/OpenHardwareMonitor.Report.txt)

moellerm, 2020-05-23:
Please try with this alpha version: https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.3.5-alpha.zip

nnchiew, 2020-05-24:
Thank you, it solves the problem.  Sent from Mail for Windows 10 From: Michael MöllerSent: Saturday, 23 May 2020 8:44 pmTo: openhardwaremonitor/openhardwaremonitorCc: nnchiew; AuthorSubject: Re: [openhardwaremonitor/openhardwaremonitor] CPU temperature not showing for i7-10875h (#1281) Please try with this alpha version: https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.3.5-alpha.zip

moellerm, 2020-05-24:
Thx for the feedback.

closed 2020-05-24 by moellerm