openhardwaremonitor/openhardwaremonitor #1286: Doesn't work on Linux

gardotd426, 2020-05-27:
The download page says that OpenHardwareMonitor works on Linux, as long as Mono with WinForms is installed in the Wineprefix.

However, it segfaults no matter what you do. I've tried both 32-bit and 64-bit Wineprefixes, and multiple versions of Wine. 

Here's the output of the crash window log:

```
Unhandled exception: page fault on read access to 0x0000fffc in 32-bit code (0x7dd552b8).
Register dump:
 CS:0023 SS:002b DS:002b ES:002b FS:0063 GS:006b
 EIP:7dd552b8 ESP:0031f090 EBP:0031f118 EFLAGS:00010a16(  R- --O I   -A-P- )
 EAX:fffffffe EBX:00000000 ECX:00000000 EDX:00007fff
 ESI:00a5b6d8 EDI:0031f0d4
Stack dump:
0x0031f090:  00000000 00000000 01082118 00000000
0x0031f0a0:  20000007 00a5b6d8 00000154 00000001
0x0031f0b0:  80000000 00000008 00000001 7bc8e255
0x0031f0c0:  0031f14c 0031f10c 0031f0f8 7bc8e255
0x0031f0d0:  00110000 00000001 010820d8 010820f8
0x0031f0e0:  00000000 00000000 0031f188 0031f110
Backtrace:
=>0 0x7dd552b8 (0x0031f118)
  1 0x7dd558fe (0x0031f1b8)
  2 0x0bd8f2dc (0x0031f248)
  3 0x0bd8f1f0 (0x0031f288)
  4 0x0bd8e651 (0x0031f2e8)
  5 0x0bd8e364 (0x0031f318)
  6 0x0bd8e194 (0x0031f368)
  7 0x0bd8e0ec (0x0031f398)
  8 0x0bd8e05c (0x0031f3c8)
  9 0x0bd6d2c8 (0x0031f448)
  10 0x0bd6b2bc (0x0031f518)
  11 0x0bb13f20 (0x0031f548)
  12 0x0b1a2fe0 (0x0031f878)
  13 0x0b1a194c (0x0031f8a8)
  14 0x003f1b9c (0x0031f8e8)
  15 0x003f19dd (0x0031f918)
  16 0x6c54c9ae EntryPoint+0xb5ad() in libmono-2.0-x86 (0x00d17a40)
0x7dd552b8: movzwl	0xfffffffe(%ebx,%edx,2),%ecx
Modules:
Module	Address			Debug info	Name (56 modules)
PE	  400000-  47e000	Deferred        openhardwaremonitor
PE	 1f30000- 23a4000	Deferred        mscorlib
PE	 2560000- 27e2000	Deferred        system
PE	 27f0000- 283e000	Deferred        system.management
PE	 2840000- 28c0000	Deferred        system.drawing
PE	 28c0000- 28c8000	Deferred        accessibility
PE	 28d0000- 2922000	Deferred        openhardwaremonitorlib
PE	 2930000- 295a000	Deferred        aga.controls
PE	 2960000- 2984000	Deferred        system.configuration
PE	 b1b0000- b1bc000	Deferred        oxyplot.windowsforms
PE	 b1c0000- b210000	Deferred        oxyplot
PE	 b650000- b95e000	Deferred        system.xml
PE	 b9d0000- bae0000	Deferred        system.core
PE	10000000-10788000	Deferred        system.windows.forms
PE	6c540000-6c960000	Export          libmono-2.0-x86
PE	7a840000-7a849000	Deferred        opengl32
PE	7b020000-7b03c000	Deferred        kernelbase
PE	7b420000-7b5f1000	Deferred        kernel32
PE	7bc30000-7bc48000	Deferred        ntdll
PE	7c920000-7c926000	Deferred        uxtheme
PE	7d3c0000-7d3c5000	Deferred        winex11
PE	7d520000-7d526000	Deferred        imm32
PE	7d550000-7d554000	Deferred        usp10
PE	7d5b0000-7d606000	Deferred        comctl32
PE	7d720000-7d74d000	Deferred        ole32
PE	7d870000-7d87e000	Deferred        gdi32
PE	7d9e0000-7dad4000	Deferred        user32
PE	7dc30000-7dc3a000	Deferred        aclui
PE	7dc70000-7dc7d000	Deferred        rpcrt4
PE	7dcf0000-7dcf7000	Deferred        sechost
PE	7dd30000-7dd3d000	Deferred        advapi32
PE	7dd90000-7dd95000	Deferred        version
PE	7ddb0000-7ddc9000	Deferred        setupapi
PE	7de30000-7de42000	Deferred        shlwapi
PE	7deb0000-7e7ac000	Deferred        shell32
PE	7e8d0000-7e8f8000	Deferred        ucrtbase
PE	7e9d0000-7e9d7000	Deferred        dbghelp
PE	7ea40000-7ea46000	Deferred        mscoree
PE	7efd0000-7efd6000	Deferred        shcore
PE	f6690000-f669a000	Deferred        wined3d
PE	f6cd0000-f6cd5000	Deferred        dxgi
PE	f6e20000-f6e2e000	Deferred        winspool
PE	f6e70000-f6e75000	Deferred        wbemprox
PE	f6f10000-f6f17000	Deferred        iphlpapi
PE	f6fe0000-f6fe8000	Deferred        propsys
PE	f7030000-f7039000	Deferred        windowscodecs
PE	f71a0000-f71a4000	Deferred        xmllite
PE	f71d0000-f71dc000	Deferred        taskschd
PE	f7230000-f7235000	Deferred        wmiutils
PE	f7860000-f7865000	Deferred        fusion
PE	f7890000-f7897000	Deferred        ws2_32
PE	f78d0000-f7950000	Deferred        winmm
PE	f7990000-f7995000	Deferred        psapi
PE	f79d0000-f79de000	Deferred        oleaut32
PE	f7b00000-f7b15000	Deferred        msvcrt
PE	f7bb0000-f7bb5000	Deferred        bcrypt
Threads:
process  tid      prio (all id:s are in hex)
00000020 (D) Z:\home\matt\dev\drmtest\GLZ-Linux-GPU-Monitoring-DRM\OpenHardwareMonitor\OpenHardwareMonitor.exe
	["Z:\home\matt\dev\drmtest\GLZ-Linux-GPU-Monitoring-DRM\OpenHardwareMonitor\OpenHardwareMonitor.exe"]
	0000025c    0
	00000234    0
	000001d0    0
	000001cc    0
	00000024    0 <==
00000030 services.exe
	00000248    0
	000001f8    0
	000001e0    0
	000001ac    0
	00000108    0
	000000fc    0
	000000f8    0
	00000038    0
	00000034    0
00000044 explorer.exe
	00000054    0
	00000048    0
00000100 winedevice.exe
	00000114    0
	00000110    0
	0000010c    0
	00000104    0
000001a4 winedevice.exe
	000001c8    0
	000001bc    0
	000001b8    0
	000001b4    0
	000001b0    0
	000001a8    0
000001d8 svchost.exe
	00000230    0
	0000022c    0
	00000228    0
	00000220    0
	0000021c    0
	000001ec    0
	000001e8    0
	000001e4    0
	000001dc    0
000001f0 rpcss.exe
	00000210    0
	00000208    0
	00000204    0
	00000200    0
	000001fc    0
	000001f4    0
00000240 winedevice.exe
	00000250    0
	0000024c    0
	00000244    0
00000260 explorer.exe
	00000270    0
	0000026c    0
	00000268    0
	00000264    0
System information:
    Wine build: wine-5.9 (Staging)
    Platform: i386 (WOW64)
    Version: Windows 7
    Host system: Linux
    Host version: 5.7.0-rc7-1-tkg-pds
```

And here's the output from the terminal:
```
=================================================================
	Native Crash Reporting
=================================================================
Got a UNKNOWN while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries
used by your application.
=================================================================

=================================================================
	Managed Stacktrace:
=================================================================
	  at <unknown> <0xffffffff>
	  at System.Security.AccessControl.NativeObjectSecurity:SetNamedSecurityInfo <0x00012>
	  at <>c__DisplayClass22_0:<InternalSet>b__0 <0x00067>
	  at System.Security.AccessControl.NativeObjectSecurity:Win32SetHelper <0x002a0>
	  at System.Security.AccessControl.NativeObjectSecurity:InternalSet <0x000fb>
	  at System.Security.AccessControl.NativeObjectSecurity:Persist <0x0004b>
	  at System.Security.AccessControl.NativeObjectSecurity:PersistModifications <0x00053>
	  at System.IO.File:SetAccessControl <0x00033>
	  at OpenHardwareMonitor.Hardware.KernelDriver:Install <0x002af>
	  at OpenHardwareMonitor.Hardware.Ring0:Open <0x00133>
	  at OpenHardwareMonitor.Hardware.Computer:Open <0x00067>
	  at OpenHardwareMonitor.GUI.MainForm:.ctor <0x0154f>
	  at OpenHardwareMonitor.GUI.MainForm:.ctor <0x0003b>
	  at OpenHardwareMonitor.Program:Main <0x0011b>
	  at System.Object:runtime_invoke_void <0x00064>
=================================================================
wine: Unhandled page fault on read access to 0000FFFC at address 7DD552B8 (thread 0024), starting debugger...
```

moellerm, 2020-05-27:
You don't need Wine to run the Open Hardware Monitor on Linux. You just need the following packages installed (from what I remember):

mono-runtime, libmono-system-windows-forms4.0-cil, libmono-system-management4.0-cil

(or mono-complete).

Then just run it with "mono OpenHardwareMonitor.exe" from the command line. The file extension is "exe", but it is not a native Windows binary, but managed CIL code (for the .NET Framework on Windows or Mono on Linux).

But keep in mind, the supported hardware is very limited on Linux. Things that should work with the current code base are: Mainboard sensors (if lm-sensors is configured correctly, OHM just acts as a user interface/interpreter for these) and ATI/AMD GPUs (if the AMD Display library is present, atiadlxx.so or atiadlxy.so).

gardotd426, 2020-05-27:
Ahhh, gotcha. 

Yeah, my Navi GPU doesn't show up, I don't know where to get those atiadlXX.so files, I'm on Arch, not Ubuntu (so those package names you gave as dependencies don't work either, only for Ubuntu/Debian and derivatives I guess), `pacman -F atiadlxx.so` has no results (`pacman -F filename` is the command that searches for which package provides a file) so it seems as though those files aren't used anymore for AMD GPUs in Linux. 

But it does work for CPU and chipset stuff:

[image]

closed 2020-05-27 by gardotd426

moellerm, 2020-05-27:
The atiadlxx.so file was provided by AMD (ATI originally), it is (or was at least) part of the official AMD GPU driver for Linux. The file is a native library that interfaces the GPU driver (internally). On the website of the AMD Display library https://gpuopen.com/adl/ they still list "Linux® in both 32- and 64-bit versions." as supported. But I'm not sure where to get the binary. The issue here https://github.com/GPUOpen-LibrariesAndSDKs/display-library/issues/10 indicates, that other have been searching for it as well. 
I do not have access to an AMD GPU on a Linux system atm, so I can't really test or search.