openhardwaremonitor/openhardwaremonitor #1287: ITE IT879XE Fan control not working

RonnieDilli, 2020-05-28:
Changes in the speed of the fans controlled by ITE IT879XE have no effect in OHM.
ITE IT8688E work's as intended.

Using Gigabyte's fan control app (System Information Viewer - SIV) all fan controllers works just fine.

Enviroment:
- Gigabyte X570 AORUS PRO WIFI (/mainboard)
- Ryzen 3900X
- Windows 10 1909
- Open Hardware Monitor v0.9.5

I tried with Gigabyte software both enabled and completely disabled, but no luck.

[20200528 OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4697469/20200528.OpenHardwareMonitor.Report.txt)

```
+- Gigabyte X570 AORUS PRO WIFI (/mainboard)
|  +- ITE IT879XE (/lpc/it879xe)
|  |  +- Fan #1         :  779.446  774.971  800.712 (/lpc/it879xe/fan/0)
|  |  +- Fan #2         :  817.191  816.203  863.171 (/lpc/it879xe/fan/1)
|  |  +- Fan Control #1 :       25       25       25 (/lpc/it879xe/control/0)
|  |  +- Fan Control #2 :       25       25       25 (/lpc/it879xe/control/1)
|  |  +- Fan Control #3 :       23       19       23 (/lpc/it879xe/control/2)
```

CSchuetz7, 2020-05-29:
For me the same problem on my Gigabyte Z390 Pro Wifi

You cant change the Fan Control on the ITE IT879XE.
The Fan Control of the ITE IT8688E works.

DarkIrata, 2020-09-24:
Got the same problem. Gigabyte X570 Master
My guess is that the FAN_PWM_CTRL_EXT_REG isn't matching up. 
If someone can give some informations on how to better check / find registers i would test it. Don't want try bruteforcing it without knowing what i am doing :D

CreeT6, 2020-12-07:
Exactly the same problem with my Gigabyte X570 AORUS PRO WIFI, 5600x. IDK if anyone wants to spend some time to fix this though.

GittyBaer, 2021-10-15:
Just to inform you:
Linux Kernel has Support for the newer Gigabyte ITE Sensorchips now:
https://github.com/t-8ch/linux-gigabyte-wmi-driver
acidanthera/bugtracker#1666

So it would be great if you can adapt it here?

(I will spread this on other cases about the same issue)