openhardwaremonitor/openhardwaremonitor #1299: Incorrect CPU temperature/plot results

RevNight, 2020-07-03:
Hello,

The issue I am facing is that for CPU Temperatures, the plot does not display the Max Value. This is concerning because the max value for CPU Core #4 and CPU Package is 73C. Yet when you look at the plot, very few of the values go above 60, and none go anywhere near 65. I am certain the plot shows the entire time length because I am resetting min/max often to test my setup.

Is the plot wrong, or is the Max Value wrong? 

Specs:

-Open HW v0.9.5
-I7-6700K (OC to 4.6ghz)
-Gigabyte Z170-HD3P
-AMD Radeon 5700XT
-16GB ram
-Win 10
[image]

martynsmall, 2020-07-22:
Hey, the reason you see a difference is because the historical values which are used by the plot don't contain every data point - instead an average value is calculated and recorded for every four. The current, min and max are updated for each point. The nature of this averaging means that any spikes which only last one or two seconds won't appear on the plot. While it would be nice to see all the data on the plot, attempting to render (and store) four times as much detail is bound to have performance implications!

If you're interested, [this](https://github.com/openhardwaremonitor/openhardwaremonitor/blob/2b9ec17583ab39e44c45c86a32123645e57e5b8e/Hardware/Sensor.cs#L186) is the block of code that deals with the averaging.

RevNight, 2020-07-22:
I see. So the whole purpose, for me, of the plot is to see which are just temperature spikes and which are sustained high temps. Obviously one is worse than the other. How can I achieve this using Open HW?

martynsmall, 2020-07-22:
You could modify the code in the block I linked to above so it starts recording and showing every data point rather than averaging them. You'd basically want to replace the contents of that `if` block with something like `AppendValue(value.Value, now);`, and then you should see every spike on the plot. However, bear in mind this was likely done for a good reason in the first place so there's the potential for the plot to become a lot slower when panning around etc.

anomalis2, 2023-07-20:
Correct me if I'm wrong, but.. If it averages 4 temp checks for 1 point on the plot, and the plot updates every 1 second, that's 4 temp checks per second for which you're seeing an average for that single second. That would mean that your max temp of 73C happened for less than a second, and not only that, but since your 1 second average of 4 temp checks was < 60, the temp for the other 3 checks during that second had to be much lower.

Maybe I'm misunderstanding, but it seems to me like if you see nothing on the plot over 60C, then anything over that would surely be considered a "spike" and nothing to worry about. Right? Perhaps like the other commenter said they did it to reduce performance issues, but it seems to me like it's also useful for your exact purpose. You want to know if your max temp is sustained or was just a spike. Well, the plot helps you with that because it would have to sustain for at least a full second for it to even show up on the plot. Therefore, what you're seeing is a max sustained (even for a second) temp of about 61C, with fraction of a second spikes (at least once) up to 73C.

anomalis2, 2023-07-20:
I wrote my last comment before bed, and now that I'm awake I'm thinking that I misunderstood. That it checks the temp once per second, and plots the average of every 4 seconds as 1 point on the plot. But still, I think it serves the basic purpose you're going for. If 1 second was at 73C but your average for 4 seconds was below 60, I'd still call that a spike.