openhardwaremonitor/openhardwaremonitor #1303: An way to add custom fan profiles?

Fround1, 2020-07-15:
im not big with coding and stuff but im wondering if its possible to add "profiles" to OHM, example, one profile is 100% fan speed on everything and the other one is 20% on every fan.

Rem0o, 2020-07-23:
Checkout https://github.com/Rem0o/FanControl.Releases, it uses OHM/LHM as a backend to do exactly that.

Fround1, 2020-07-23:
> 
> 
> Checkout https://github.com/Rem0o/FanControl.Releases, it uses OHM/LHM as a backend to do exactly that.

it doesnt seem like it changes my fans, i used the automatic config and it just sets the fan to default (58%) and doesnt go up or down even if i stress the pc, i tried closing OHM also but nothing worked.

dvdvideo1234, 2020-07-27:
@Fround1 

Try [`SpeedFan`](https://www.almico.com/speedfan452.exe) It appears to work for me. Requires a little bit control theory to set it up but nothing an IT guy can't handle ( I hope ). If you are still having trouble with it, I've made videos how to setup it with [NH-D15S](https://noctua.at/en/nh-d15s). For [part 1 click here](https://www.youtube.com/watch?v=ny6kvzAt1eI), and [for part 2 here](https://www.youtube.com/watch?v=Y1Hkf3KkRLE).