openhardwaremonitor/openhardwaremonitor #1309: SSD not being read?

farp96, 2020-08-01:
I got a new Samsung 970 Evo Plus SSD 1TB. With HW monitor it sees my SSD and temps fine. But using open hardware monitor it doesn't see it. Is there a reason why it doesn't read temps of SSD's? It's installed using NVME.

Crownzhu, 2020-08-08:
I got the same problem. How to solve ?

Dude-sks, 2020-08-14:
Same problem here. Temperature does not update, stuck on 30C (i was able to read temp correctly with Samsung Magician). Samsung 970 Evo SSD 250gb, Gigabyte Z370 D3 15f, Win10 19041.

vamsivadrevu, 2020-09-08:
Yes Samsung 970 Evo SSD not being identified. Shows as Generic Hard Disk. Should it's identifier be added to the code? I'm not exactly sure how to do this. Please help me understand the code so that I can contribute better.

necris77, 2020-09-16:
Kingston A2000 1Tb is not identified either, it shows as generic hard disk and no temperatures tab

botekp, 2020-09-17:
No temperatures shown of NVME and USB drives either...

ChristopherW, 2020-10-15:
Likewise, on my Asus Maximus Hero XI (Z390 platform, Nuvoton NCT6798D) a brand new Samsung 970 Evo Plus is not read, parsed only as "Generic Hard Disk". An Intel 750 Series PCIe SSD is also read as "Generic Hard Disk". Only the Western Digital spinning disks show temperatures. Issue present on 0.9.2 and 0.9.5. Have also submitted a report through the app.

donpin1, 2021-02-19:
Agree, please add NVMe drives and their temperatures. It would be great if you also had a tab that monitored SMART attributes. Nevertheless, this is easily the best program for me that I have come across. Well done.
[image]