openhardwaremonitor/openhardwaremonitor #1313: Z490 Msi Unify + 10700k, cpu speed seems to ignore buspeed

SteadVex, 2020-08-05:
Not sure if its just a Z490 thing, but cpu speeds seem to ignore buspeed, or blkclk, not sure of its proper name

seems to always get speed for me based on 99.75 approximately no mater what it is to in the bios

only open hardware monitor reports it this way, hw monitor, hwinfo, cpu-z intel tuning etc all report the correct speed

also have no voltages for anything CPU/Mobo related, although putting that down to fairly new release

Also, this is great software, really like the desktop gadget :)

nuno-aj-aniceto, 2020-08-11:
Hi,
I also have similar data.
Motherboard: MSI Z490 Unify
Processor: Intel 10700k (10th Gen)

Follow a printscreen

[image]

jordigil, 2020-08-13:
Same here for a MSI B550 Tomahawk + Ryzen 3900x processor show info but the motherboard is empty.