openhardwaremonitor/openhardwaremonitor #1314: OpenHardwareMonitor.config Increases In Size After Exit

dexterslab2013, 2020-08-08:
Every time you exit the OpenHardwareMonitor.config file increases in size.

My config file was over 3Mb when i noticed.

Using 0.9.5

Zud71, 2020-08-17:
Each time the configuration is saved (when the program is closed), data from the sensors is accumulated. Why this is done is not yet clear.
I can change this part of the code, but ...

megaslowpoke, 2021-03-28:
> I can change this part of the code, but ...

It would be nice.
Maybe give the user the ability to limit the story, and not only in time, but also in size. Or add as an option: delete history on exit, then there will be nothing superfluous in the config upon completion of the work.

In the meantime, you have to use a crutch so as not to lose settings due to too large a config, i.e. before starting, I simply remove everything unnecessary from the xml:

```
type OpenHardwareMonitor.config | findstr /i "version configuration appSettings tray traycolor startMinMenuItem minCloseMenuItem appSettings configuration">>OpenHardwareMonitor.config.new 2>nul
erase /f /q OpenHardwareMonitor.config 1> nul
rename OpenHardwareMonitor.config.new OpenHardwareMonitor.config
```
"2>nul" is to hide how findstr is swearing at too long lines of history.

UnclWish, 2022-10-11:
Problem is still exists. It must be the way to limit this sensors values... And it's need an option to clear them...