openhardwaremonitor/openhardwaremonitor #1319: System.IO.FileNotFoundException: "Could not load file or assembly"

Just-Ch33s3, 2020-08-17:
Hi, 
im currently trying to read out CPU values with Visual Studio in C# following this tutorial I've found : http://www.lattepanda.com/topic-f11t3004.html
Currently im getting this error, which possibly states that ;
###
System.IO.FileNotFoundException: "Could not load file or assembly 'System.Management, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'.
###
Here's the error as screenshot
[image]
Does someone might have a clue what I am missing here ?
Cheers, have a good day!

chungggg, 2020-08-24:
I had same issue when runing on .netcore framework, then I installed System.Management from Nuget, issue is resolved but I received new issue No method: System.Threading.Mutex
I think project is write for .Net Framework, not .netcore. I'm trying to edit the code for working with .netcore

Just-Ch33s3, 2020-08-24:
Nice thanks for your answer, would be awesome if you could report back after you've found a solution! :)

chungggg, 2020-08-25:
Ok. I'm resolved issue when move to dotnet core framework instead of .Net Framework
1. Change OpenHardwareMonitorLib to dotnet core
2. Install dependencies packages: System.IO.FileSystem.AccessControl, System.Management, System.Reflection.Emit
3. Change some source code:
On Ring0.cs:
`isaBusMutex = Mutex.OpenExisting(isaMutexName, MutexRights.Synchronize);` to `isaBusMutex = Mutex.OpenExisting(isaMutexName);`
`pciBusMutex = Mutex.OpenExisting(pciMutexName, MutexRights.Synchronize);` to `pciBusMutex = Mutex.OpenExisting(pciMutexName);`
On KernelDriver.cs:
`FileSecurity fileSecurity = File.GetAccessControl(@"\\.\" + id);` to `FileInfo fileInfo = new FileInfo(@"\\.\" + id);
                FileSecurity fileSecurity = fileInfo.GetAccessControl();`
`File.SetAccessControl(@"\\.\" + id, fileSecurity);` to `fileInfo.SetAccessControl(fileSecurity);`
On PInvokeDelegateFactory.cs:
`      AppDomain.CurrentDomain.DefineDynamicAssembly(
        new AssemblyName("PInvokeDelegateFactoryInternalAssembly"),
        AssemblyBuilderAccess.Run).DefineDynamicModule(
        "PInvokeDelegateFactoryInternalModule");` to `AssemblyBuilder.DefineDynamicAssembly(
        new AssemblyName("PInvokeDelegateFactoryInternalAssembly"),
        AssemblyBuilderAccess.Run).DefineDynamicModule(
        "PInvokeDelegateFactoryInternalModule");`