openhardwaremonitor/openhardwaremonitor #1322: CPU temperatures of 204.9 degrees C!

MrYossu, 2020-08-20:
Hello,

As my PC was getting quite warm last week, I pointed a fan at it, and left OHM running the whole time so I could keep an eye on the CPU and GPU temperatures. Earlier today (a pleasant, but especially warm day, and I have window open), I noticed that one of the cores was reported as running at 204.9 degrees C. It's now 20:30, and has cooled down somewhat, and now three of the CPU cores are showing that exact temperature. The max temperatures claim that almost all of the cores have hit that temperature at some point.

Apart from the obvious fact that I would probably have noticed if my PC had reached anywhere near that temperature (😋), it is highly suspicious that all of these temperatures are exactly the same.

I have attached a report from OHM, and a screenshot of the CPU temperatures.

Anyone any comments? Thanks

[image]

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5105113/OpenHardwareMonitor.Report.txt)

shrddr, 2021-07-11:
related? #1277

MrYossu, 2021-07-11:
@shrddr Thanks, I just got the latest and all looks good.

closed 2021-07-11 by MrYossu