openhardwaremonitor/openhardwaremonitor #1325: Program shows fantom "Generic Hard Drive"

inliquid, 2020-08-29:
Win10 Pro
v0.9.5
There are 2 drives in the system :
1. System NVMe SSD
2. Additional HDD

Program shows 3 drives including some mysterious "Generic Hard Drive" with no attributes at all:
[image]

Here is for example how it looks like in HWMonitor:
[image]

ChelOis, 2020-09-29:
I have a somewhat similar problem. I reported it as Issue #1345.

kevinnio, 2020-11-15:
I'm seeing the same "Generic Hard Disk" on my system. Not a major issue, but weird nonetheless.

donpin1, 2021-02-19:
Agree, please add NVMe drives and their temperatures. It would be great if you also had a tab that monitored SMART attributes. Nevertheless, this is easily the best program for me that I have come across. Well done.
[image]