openhardwaremonitor/openhardwaremonitor #1326: ASUS Prime B450M-A IT8655E No FANS Shown

MadTomT, 2020-08-31:
As per subject, I've got no fans at all showing when I run open hardware monitor.
This include motherboard or CPU fans.

Other apps work, so I know it's not a sensor issue.
Anyone any idea how to resolve this ?

Open Hardware monitor 0.9.5
Thanks

jordigil, 2020-08-31:
Same version but my motherboard is a MSI MAG B550 TOMAHAWK (MS-7C91)

Motherboard category is empty only showing the name of the MB.

MadTomT, 2020-08-31:
Would be good to get this working if any one has any ideas.. thanks

weretheone, 2020-09-08:
Hello, I have the same issue with Asus TUF B450 with IT8665E. It shows nothing in open hardware monitor. What is strange that I ended up installing CPU-Z, just to check other things, then I noticed that a lot of extra details popped up. So now if I start CPU-Z then OpenHardware monitor it shows everything correctly. All the motherboard temperature sensors along with the Fan speeds etc.

MadTomT, 2020-09-09:
I've tested this and I get the same. With CPU-Z running the fan and other temps are shown.
Interestingly the temps show are about 5-10 degrees more than Ryzen Master

weretheone, 2020-09-09:
Yeah so this is definately some kind of bug, I tried to track down and comment on the last commit which changed the file responsible for this: https://github.com/openhardwaremonitor/openhardwaremonitor/commit/7bf4f655d2c3ad9b660b6703c632b1d5694778ca

As I understand multiple chip was added to run with a different method, let's wait and see if the author has time and resource to follow up on this.

atmosfar, 2020-12-19:
Same issue. I can see that the CPU Fan speed/% appears in CPUID's HWmon.

pablorrrrr, 2020-12-20:
Same "issue" here - after starting CPUIDs HWmon - and then starting OHW all fans show up. (but not without starting CPUID's)

Fannon, 2021-09-05:
Same here with:

```
+- MSI B550-A PRO (MS-7C56) (/mainboard)
```

```
Mainboard

SMBIOS Version: 2.8

BIOS Vendor: American Megatrends International, LLC.
BIOS Version: A.82

System Manufacturer: Micro-Star International Co., Ltd.
System Name: MS-7C56
System Version: 2.0

Mainboard Manufacturer: Micro-Star International Co., Ltd.
Mainboard Name: B550-A PRO (MS-7C56)
Mainboard Version: 2.0

Processor Manufacturer: Advanced Micro Devices, Inc.
Processor Version: AMD Ryzen 7 5800X 8-Core Processor
Processor Core Count: 8
Processor Core Enabled: 8
Processor Thread Count: 16
Processor External Clock: 100 Mhz
```