openhardwaremonitor/openhardwaremonitor #1332: Crucial M.2 NVME 1TB SSD (OS), NO INFO DISPLAYED.

BCC1, 2020-09-11:
Crucial M.2 NVMe SSD (OS) info is  displayed only as Generic Hard Disk and shows only Used Space percentage. Secondary HDD (storage), attached USB drive and other hardware info is displayed as expected.

Phoenix-, 2020-09-25:
Same for Samsung 970 Evo, with samsing Nvme driver(if that matters)
OHM 0.9.5
Windows 10 x64 2004, all updates installed.

maxima120, 2020-09-27:
Would like to see temperature readings for Gigabyte Aorus NVMe also

donpin1, 2021-02-19:
Agree, please add NVMe drives and their temperatures. It would be great if you also had a tab that monitored SMART attributes. Nevertheless, this is easily the best program for me that I have come across. Well done.
[image]