openhardwaremonitor/openhardwaremonitor #1338: [Asus TUF B450M] Fans are not appearing

RevNight, 2020-09-19:
Hello,

I have two PCs in my house, one gaming (i7-6700K, Gigabyte Z170) and one htpc (Ryzen 3600, Asus TUF B450M). The gaming one runs OpenHW perfectly and I've never had any issues with it. The htpc, even after installing the most recent OpenHW of 0.9.5, is not showing the Case/CPU fans. I have selected 'File-Hardware-Mainboard' and also 'View-Show Hidden Sensors', but there is nothing under the motherboard for me to select. I'm pretty sure I have the most recent mobo drivers as this is pretty brand new. I can adjust the PWM fans just fine in Bios.

What could be causing these not to show? I would prefer not to flash the bios.

Fans:
-Intake: Noctua NF-A14
-Exhaust: Noctua NF-S12A
-Cpu: Noctua U12S

Attached:
-OpenHW screenshot
-OpenHW report

[image]
[OpenHardwareMonitor.Report. - Asus TUF B450M.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5250067/OpenHardwareMonitor.Report.-.Asus.TUF.B450M.txt)

RevNight, 2020-10-09:
Original post edited with fan controller

ThePegass, 2020-10-10:
Hi, here is the same with the same S I/O chip for B450M PLUS board. 
[image]

RevNight, 2020-10-12:
Looks like I'm not the only one. IT8655E support is still needed. Thanks!