openhardwaremonitor/openhardwaremonitor #1351: Additional Information : VCore and VTT

DigidogIan, 2020-10-12:
Similar to issues 1238 and 1305, Vcore and VTT voltages exactly one half of expected value on Ryzen 5 3600 / ASUS ROG STRIX B550-F Gaming.  Win 10 Pro V. 2004 Build 19041.508.

[image]

I hope this info helps toward issue. Please forgive me if what I have done is bad etiquette here. I am used to forums.

Ian.

xcasxcursex, 2021-01-26:
#1238 #1305

Hi Ian, 
Just so you know, you can link this issue and those, by mentioning the other issues with the # prefix, as above.

I've a B550-E board on the way - very similar to this one; I'll test and report back.

ijinkli, 2021-02-14:
I came here to find why there's no VCore measurement on Nuvoton NCT6791D. Can confirm this issue
[image]
should be an easy fix? Thanks for this program :D

xcasxcursex, 2021-02-18:
Confirmed this issue exists both here and on librehardwaremonitor. Since this project is [dead](https://github.com/openhardwaremonitor/openhardwaremonitor/graphs/commit-activity), AGAIN, I'll be leaving it permanently as of right now, and reporting this bug over there.

ijinkli, 2021-02-18:
Hi I was looking at maybe fixing the code myself but didn't realise this
project was DEAD, thanks for the info, I'll have a look at the link