openhardwaremonitor/openhardwaremonitor #1355: i7-10750H clocks and temps aren't showing

HunamX, 2020-10-19:
[image]

Soraldor, 2020-10-20:
Can you provide a report for this cpu? (File -> Save Report and attach it to this issue, be sure to execute the monitor as an administrator) 
I just wanted to open an issue by myself for an Intel(R) Core(TM) i7-1065G7 CPU (Report attached, started as an administrator) - maybe it's the same problem so I hope it's okay if I add my issue to this. 
[admin_OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5407848/admin_OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2020-10-20:
@Soraldor Did you check the Log? You got a Problem with the Ring0 Service! I would recommend, that you fix that! (redownload from a TRUSTED Source! If it was not from here!)

If that does not help, it could be, that windows is blocking this "unsigned" driver.
There was an Option to disable that... Think google will lead you!

HunamX, 2020-10-20:
This is work PC, So after running it as admin it installed OpenHardwareMonitorLib.sys and all is well.
[image]

closed 2020-12-27 by moellerm