openhardwaremonitor/openhardwaremonitor #1357: OHM app reports GPU power, C# lib does not

fyrewurx, 2020-10-26:
Hey there, as per title the GPU power is shown in the app but it's missing from the lib. For both the version is 0.9.5.0 . Adding relevant parts of the report, full report attached. Any ideas how to get it in C#?

[libreport.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5438193/libreport.txt)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5438194/OpenHardwareMonitor.Report.txt)

Report from lib:

+- NVIDIA GeForce GTX 960 (/nvidiagpu/0)
|  +- GPU Core       :  1227.26  1227.26  1227.26 (/nvidiagpu/0/clock/0)
|  +- GPU Memory     :     3600     3600     3600 (/nvidiagpu/0/clock/1)
|  +- GPU Shader     :  2454.52  2454.52  2454.52 (/nvidiagpu/0/clock/2)
|  +- GPU Core       :       36       36       36 (/nvidiagpu/0/temperature/0)
|  +- GPU Core       :        0        0        0 (/nvidiagpu/0/load/0)
|  +- GPU Frame Buffer :        1        1        1 (/nvidiagpu/0/load/1)
|  +- GPU Video Engine :        0        0        0 (/nvidiagpu/0/load/2)
|  +- GPU Bus Interface :        0        0        0 (/nvidiagpu/0/load/3)
|  +- GPU Memory     :  45.5273  45.5273  45.5273 (/nvidiagpu/0/load/4)
|  +- GPU            :        0        0        0 (/nvidiagpu/0/fan/0)
|  +- GPU Fan        :        4        4        4 (/nvidiagpu/0/control/0)
|  +- GPU Memory Free :   1115.6   1115.6   1115.6 (/nvidiagpu/0/smalldata/1)
|  +- GPU Memory Used :  932.398  932.398  932.398 (/nvidiagpu/0/smalldata/2)
|  +- GPU Memory Total :     2048     2048     2048 (/nvidiagpu/0/smalldata/3)

Report from app:

+- NVIDIA GeForce GTX 960 (/nvidiagpu/0)
|  +- GPU Core       :      135      135      135 (/nvidiagpu/0/clock/0)
|  +- GPU Memory     :      405      405      405 (/nvidiagpu/0/clock/1)
|  +- GPU Shader     :      270      270      270 (/nvidiagpu/0/clock/2)
|  +- GPU Core       :       35       35       35 (/nvidiagpu/0/temperature/0)
|  +- GPU Core       :        0        0       21 (/nvidiagpu/0/load/0)
|  +- GPU Frame Buffer :        5        5       19 (/nvidiagpu/0/load/1)
|  +- GPU Video Engine :        0        0        0 (/nvidiagpu/0/load/2)
|  +- GPU Bus Interface :        0        0        1 (/nvidiagpu/0/load/3)
|  +- GPU Memory     :  45.8324  45.6249  45.9301 (/nvidiagpu/0/load/4)
|  +- GPU            :        0        0        0 (/nvidiagpu/0/fan/0)
|  +- GPU Fan        :        4        4        4 (/nvidiagpu/0/control/0)
|  +- GPU Power      :   10.148    10.05   11.307 (/nvidiagpu/0/power/0)
|  +- GPU Memory Free :  1109.35  1107.35   1113.6 (/nvidiagpu/0/smalldata/1)
|  +- GPU Memory Used :  938.648  934.398  940.648 (/nvidiagpu/0/smalldata/2)
|  +- GPU Memory Total :     2048     2048     2048 (/nvidiagpu/0/smalldata/3)
|  +- GPU PCIE Rx    :        0        0  376.953 (/nvidiagpu/0/throughput/0)
|  +- GPU PCIE Tx    :        0        0  1.95313 (/nvidiagpu/0/throughput/1)

hsinyu-chen, 2020-11-06:
same for EVGA 3080

MerlinTheProgramist, 2020-11-18:
Same for nvidia 2070

abdonkov, 2020-11-25:
Had the same problem and found a solution over at the LibreHardwareMonitor github page (a fork of this project which is actively developed still)...

So in my project Build Platform Type was set to 'Any CPU' with the 'Prefer 32-bit' option checked. So unchecking this option, or changing the platform to 'x64' instead fixed the problem. It looks like the sensor readings doesn't work on 32bit.

fyrewurx, 2020-11-26:
@abdonkov thank you very much, that was it, now i have all the info I need. Cheers!

closed 2020-11-26 by fyrewurx