openhardwaremonitor/openhardwaremonitor #1359: Plot massive zoom Issues

Eiion, 2020-11-05:
I'm having repeatedly issues with zooming the plot and while I previously was able reset view to a certain time period, I cannot reset temperature range. So you can (accidentally) zoom in temperature range to infinity - but good luck getting it back to about 0-100°C. It's impossible.
Currently I'm not even getting the axes displayed in the graph anymore. Closing down Open Hardware Monitor and opening it back up several times is of no help at all. I'm at a point where the OPM is unusable and I'll have to try it with uninstalling and reinstalling it.

That plot is in urgent need to be properly reset to standard ranges and considering how people have asked for this over 5 years ago already I don't get how this couldn't be implemented by now. Please take care of this.

moellerm, 2020-12-27:
There are several ways to manipulate the plots:

- Left click and drag in the plot area to move the range on both axes (of the corresponding subplot).
- Left double click in the plot area to auto-fit (reset) all data on both axes (of the corresponding subplot).
- Mouse wheel scroll while the mouse is in the plot area to zoom both axes (of the corresponding subplot).

- Left click and drag on an axis (outside the plot area) to move the range of this axis only.
- Left double click on an axis (outside the plot area) to auto-fit (reset) the axis range to the data.
- Mouse wheel scroll while the mouse is on an axis (outside the plot area) to zoom the axis range.

- Right click on the plot -> Time Window -> Select a predefined time window.

Normally this should cover all needs. But it might be a bit difficult to "discover" this. Maybe I could add an additional command to the context menu to reset all vertical axes?

Eiion, 2020-12-28:
Those options are all well known to me - still they are of virtually no help if you got lost in the emty space of the graph from scrolling too quick. Hence my request.

shrddr, 2021-07-11:
Yea context menu > "reset Y axis" entry is a common way to do this