openhardwaremonitor/openhardwaremonitor #1361: Lib in C++

samjviana, 2020-11-07:
Is there a way where i can use/import the .dll in an c++ code?

koogar, 2021-01-18:
yo might get some clues here
https://github.com/koogar

samjviana, 2021-01-19:
As i could see, you have two applications working kinda together, one in C# and the other on C++, and the C# one that makes use of the OHM lib. By the way, wonderful work you're doing.
The thing is, i sorta wrote an API running with an C# application  (similar to "HardwareSerialMonitor"), that way i could get the information i need over an HTTP Request. But i was working on an parallel project with an C++ application and thought if i could directly call the functions from the OHM lib inside the C++ without interfacing with an C# application.