openhardwaremonitor/openhardwaremonitor #1364: M.2 NVMe SSD - No info displayed

billgat, 2020-11-10:


yunyuyuan, 2020-12-07:
Me too.Hope to add this feature earlier

MJAvH, 2021-01-05:
On MSI X470 GAMING PRO MAX (MS-7B79) also no info from M.2 NVMe SSD drives.

lprocx, 2021-01-28:
A little while ago i had the same problem. Fixed it myself in a fork. Haven't had time to follow up on it, but since more people are having problems I might make time for it soon.

Can you you check my fork if it solves your issue? Link: https://github.com/lprocx/openhardwaremonitor

donpin1, 2021-02-19:
Agree, please add NVMe drives and their temperatures. It would be great if you also had a tab that monitored SMART attributes. Nevertheless, this is easily the best program for me that I have come across. Well done.
[image]