openhardwaremonitor/openhardwaremonitor #1368: More sensors for Nuvoton NCT6798D

holgerpieta, 2020-11-19:
The Nuvoton NCT6798D on my Asus TUF Gaming B550M-Plus (Wi-Fi) appears to be supporting more temperature sensors than are currently read out in OHM.
They are visible both in BIOS and in HWiNFO and show reasonable values, so they seem to be really there and not just unconnected ports. Which is a bit of a shame, because the "CPU temperature" used by the BIOS to control the fans is one of the sensors missing.

I had a look at how support for the NCT6798D was added in commit b4485c80d79476efcfc296b50125b9d7728560a5 and it looks like it was just enabled as a supported sensor in the generic NCT677X support. Maybe we need a new support for NCT679X chips with at least 3 more temperature sensors.
Also I'm a bit surprised to see 7 temperature sensors configured in NCT677X.cs, but only 6 are visible (none hidden) when I run the program. Is there some other part in the code that removes sensors again if the values are bogus or something like that?

I'm willing to dig into that, but I'm missing some fundamental build instructions. I gathered I need Visual Studio. VSCode is not enough?

cmeu, 2020-12-31:
I have the same chip in an ASUS TUF Gaming X570-Plus m/b -- Could it be as simple as the missing temperature for the CPU appears under the CPU hardware section instead?

holgerpieta, 2021-01-02:
After finding LibreHardwareMonitor, a somewhat distant fork of this project, which has got some features more and some less, I decided that LHM was closer to the target than OHM so I worked on that and added all sensors for that chip (with some significant changes to how the code works to clean things up) and also configured it correctly for my board.
The chip should work even for your board, but the sensors will have generic names (Temp 1...).
If you are willing to move to LHM, we can help you over there to get your board configured correctly.
If someone from OHM is willing to pull over my changes from LHM, I can also help here to get the board configured correctly. But in the moment I do not have the motivation to bring the changes over here myself, because LHM works for me.

holgerpieta, 2021-04-14:
Going to close the issue because I switched to LHM for good.

closed 2021-04-14 by holgerpieta