openhardwaremonitor/openhardwaremonitor #1369: Software is completely broken under Linux

Yuri6037, 2020-11-21:
Absolutely no information is displayed! The software isn't even able to know that I'm running NVIDIA GTX 650 Ti.
[image]

JohnFajardo, 2021-07-12:
Same issue here
[image]

JohnFajardo, 2021-07-12:
Same issue here. 
[image]

K4LCIFER, 2022-07-29:
Same issue for me. It appears to be poorly designed for linux, as well; it is barely legible, and incredibly non user friendly.

[image]

OS: Arch Linux
Desktop Environment: Gnome 42.3.1 (Wayland)