openhardwaremonitor/openhardwaremonitor #1374: AMD Ryzen temperatures monitoring

ATRedline, 2020-12-07:
I need to get temperatures of CPU and GPU throw OpenHardwareMonitorLib, and i made the simple code:
[image]

 It works well on intel CPUs, but i asked friend to test it, and somewhy, on his ryzen processor, my code is not seen temperatures of CPU:
[image]

 but in the OpenHardwareMonitor app this sensor exists:
[image]

 How i can get data from Ryzen termal sensor??

ATRedline, 2020-12-07:
ok. The solution was to run this app with admin rights...

moellerm, 2020-12-27:
Admin rights are required for certain sensors, not much we can do atm.

closed 2020-12-27 by moellerm