openhardwaremonitor/openhardwaremonitor #1376: NVIDIA driver 460.79 blocks reading GPU power consumption?

shocker-0x15, 2020-12-12:
Hi,
Since I updated the NVIDIA driver to 460.79, Open Hardware Monitor seems not able to show GPU power consumption.
I searched to see others meeting the same issue, there seems:
https://forums.guru3d.com/threads/nvidia-geforce-460-79-driver-download-and-discussion.435609/page-4

Open Hardware Monitor 0.9.5
Windows 10 20H2
Core i9-9900K
Geforce RTX 3080
NVIDIA Driver 460.79

Thanks,
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5683601/OpenHardwareMonitor.Report.txt)

MerlinStWahgwaan, 2020-12-14:
I am having the same issue with OHM...

almost same specs too...

i9-9900K
RTX2080
Nvidia Driver 460.79

Worked fine yesterday, did the update last night, and now today, no GPU Power readings....

Zefod, 2020-12-15:
Same here;  it's probably because of the latest NVIDIA driver... 

Windows 10 20H2
Core  i7-4790
Geforce RTX 2080 Super
NVIDIA Driver 460.79

woahwotsdis, 2020-12-15:
Having the same issue but with a 10 series card and tried both of the NVIDIA driver updates (460.79 & 460.89)

Windows 10 20H2
R5-1600AF 
Geforce RTX 1080
NVIDIA Driver 460.89

taleshape, 2020-12-15:
I have the same issue.

Windows 10 Pro 20H2
AMD Ryzen 9 3900X
GeForce RTX 2060 SUPER
NVIDIA Drivers 460.79 & 460.89

anjisZhou, 2020-12-17:
`/*
 
  This Source Code Form is subject to the terms of the Mozilla Public
  License, v. 2.0. If a copy of the MPL was not distributed with this
  file, You can obtain one at http://mozilla.org/MPL/2.0/.
 
  Copyright (C) 2020 Michael Möller <mmoeller@openhardwaremonitor.org>
 
*/

using System;
using System.Runtime.InteropServices;

namespace OpenHardwareMonitor.Hardware.Nvidia {
  internal class NVML {
    [UnmanagedFunctionPointer(CallingConvention.Cdecl)]
    private delegate NvmlReturn nvmlInitWithFlagsDelegate(uint flag);
    private static readonly nvmlInitWithFlagsDelegate nvmlInitWithFlags;

    [UnmanagedFunctionPointer(CallingConvention.Cdecl)]
    private delegate NvmlReturn nvmlInitDelegate();
    private static readonly nvmlInitDelegate nvmlInit;

    [UnmanagedFunctionPointer(CallingConvention.Cdecl)]
    private delegate NvmlReturn nvmlInit_v2Delegate();
    private static readonly nvmlInit_v2Delegate nvmlInit_v2;
      
    [UnmanagedFunctionPointer(CallingConvention.Cdecl)]
    public delegate NvmlReturn nvmlShutdownDelegate();
    public static readonly nvmlShutdownDelegate NvmlShutdown;

    public delegate NvmlReturn nvmlDeviceGetHandleByPciBusIdDelegate(
      string pciBusId, out NvmlDevice device);
    public static readonly nvmlDeviceGetHandleByPciBusIdDelegate
      NvmlDeviceGetHandleByPciBusId;

    [UnmanagedFunctionPointer(CallingConvention.Cdecl)]
    public delegate NvmlReturn nvmlDeviceGetPowerUsageDelegate(
      NvmlDevice device, out int power);
    public static readonly nvmlDeviceGetPowerUsageDelegate 
      NvmlDeviceGetPowerUsage;

    [UnmanagedFunctionPointer(CallingConvention.Cdecl)]
    public delegate NvmlReturn nvmlDeviceGetPcieThroughputDelegate(
      NvmlDevice devicee, NvmlPcieUtilCounter counter, out uint value);
    public static readonly nvmlDeviceGetPcieThroughputDelegate
      NvmlDeviceGetPcieThroughput;

    public static NvmlReturn NvmlInit() {
      try {
        var result = nvmlInitWithFlags(1);
        //var result = nvmlInit_v2();
        initialized = result == NvmlReturn.Success || result == NvmlReturn.ErrorAlreadyInitialized;
        return result;
      } catch {
        try {
          var result = nvmlInit();
          initialized = result == NvmlReturn.Success;
          return result;
        } catch {
          return NvmlReturn.ErrorLibraryNotFound;
        }
      }
    }

    private static string GetDllName() {
      if (OperatingSystem.IsUnix)
        return "libnvidia-ml.so";
      else
        return "nvml.dll";
[... 155 lines cut ...]
      /// Insufficient memory
      /// </summary>
      ErrorMemory = 20,

      /// <summary>
      /// No data
      /// </summary>
      ErrorNoData = 21,

      /// <summary>
      /// The requested vgpu operation is not available on target device, 
      /// becasue ECC is enabled
      /// </summary>
      ErrorvGpuEccNotSupported = 22,

      /// <summary>
      /// An internal driver error occurred
      /// </summary>
      ErrorUnknown = 999
    };

  }
}
`
To replace NVML.cs, it can be solved

MerlinStWahgwaan, 2020-12-21:
Working HOTFIX from Nvidia....
Must be installed manually, but I can fonfirm, it works!

https://www.nvidia.com/en-us/geforce/forums/game-ready-drivers/13/421696/announcing-geforce-hotfix-driver-46097-released-12/

rogueblue42, 2020-12-23:
> 
> 
> Working HOTFIX from Nvidia....
> Must be installed manually, but I can fonfirm, it works!
> 
> https://www.nvidia.com/en-us/geforce/forums/game-ready-drivers/13/421696/announcing-geforce-hotfix-driver-46097-released-12/

Thank you!!! This got the power back :-)

GettnBetter, 2021-01-21:
I did not have the same experience.  The hotfix and all future releases of the NVIDIA drivers seem to block this ability.  I don't get any power readings from my 2070 Super anymore.

MerlinStWahgwaan, 2021-01-21:
Thats strange... I also have a 2070 Super in my desktop, and its been
working fine since the hotfix and the new 461.08 driver....

Make sure OHM is fully closed when you install the driver, or restart it
after installing the driver.

Other than that, I would hope someone else has an idea for you because it
should work now...

MerlinStWahgwaan, 2021-01-21:
Also, make check in OHM directly, and make sure its not just your gadget not showing the readings, if thats what your using... it might recognize it as a new reading and need to be re added to the gadget

rogueblue42, 2021-01-21:
I can confirm that the latest Nvidia release 461.09 still works for my 3080, just as the hot fix did.

MerlinStWahgwaan, 2021-01-22:
> I did not have the same experience. The hotfix and all future releases of the NVIDIA drivers seem to block this ability. I don't get any power readings from my 2070 Super anymore.

Any luck?

GettnBetter, 2021-01-22:
Thank you for the suggestions!   Very much appreciated.

I closed OHM and then manually installed the latest NVIDIA drivers and all is working.

I have power again woot!   The PCIe bandwidth came back too.

MerlinStWahgwaan, 2021-01-22:
Awsome. Glad to hear it!
Have fun and enjoy!

shocker-0x15, 2021-01-23:
I had been not able to get back power reading even if reinstall the driver.
But I tried again to install the latest driver 461.09 while ensuring closing the OHM before install and selected clean install to make sure. I've confirmed the OHM now has power reading.

I'm closing this thread.
Thanks for cooperation!

closed 2021-01-23 by shocker-0x15

rogueblue42, 2021-02-06:
Not sure if I can still respond to this but this issue is back for me 461.09 even though it worked before. I used DDU to do a clean wipe of the graphic driver and reinstalled 461.09 and the GPU power is no longer displayed.

MerlinStWahgwaan, 2021-02-09:
> Not sure if I can still respond to this but this issue is back for me 461.09 even though it worked before. I used DDU to do a clean wipe of the graphic driver and reinstalled 461.09 and the GPU power is no longer displayed.

Have you followed the above reinstall suggestion, make sure OHM is fully closed, fresh reinstall of drivers, then reopen OHM?
That seems to have fixed everyone's issues thus far.
Doesn't hurt to add in a reboot before relaunching OHM, just to ensure the new drivers are in proper use....

I have not had an issues with the current drivers, on my 2080 or my 2070 Super computers. In fact, I had a separate issue wirh other hardware just last week, that ended up forcing me to do a complete factory restore on my Alienware with the 2080, back to fresh windows, everything... stupid but I had no choice because I couldn't solve my issue... and since updating back to most up-to-date, still no issues with OHM or power readings from GPU, except in Intel XTU, it seems unable to grab proper GPU data since the 460.79 update that started this issue for everyone on OHM...... which I've raised to Nvidia myself as an issue but they don't seem in too much of a rush to fix it....

If you try the fully closed OHM and GPU driver reinstall and that still fails, I would create a separate thread for your current issue with the current driver number listed, maybe someone will have other ideas for you....

MerlinStWahgwaan, 2021-02-09:
> Not sure if I can still respond to this but this issue is back for me 461.09 even though it worked before. I used DDU to do a clean wipe of the graphic driver and reinstalled 461.09 and the GPU power is no longer displayed.

Double check your driver with Nvidia website....
I just double checked mine because your version sounded odd to me....

My current driver on both my devices, is 461.40

You said you have 461.09.

You should close OHM, and update to the newest driver manually if GeForce Experience is not finding it automatically for you.... but check the Nvidia website for your specific device and make sure your on the most recent driver forsure before relaunching OHM

rogueblue42, 2021-03-07:
> Double check your driver with Nvidia website....
> I just double checked mine because your version sounded odd to me....
>
> My current driver on both my devices, is 461.40
>
> You said you have 461.09. [...]

Thank you, this worked. I updated Nvidia directly from 461.09 to 461.72. I closed OHM, did a clean install, and reboot. OHM again showed power.