openhardwaremonitor/openhardwaremonitor #1378: dark theme + hour axis

Jhnosmith, 2020-12-18:
hi, this soft is awesome, 
but I would suggest for the next version to have a dark theme,
and about the time windows : to show actual time for x axis, instead of time passed. For example, to see directly on the graphic at what hour was an irregular power consumption in the night. 
and one more time thank you and congratulation for this very good tool.

GeekBoyMN, 2020-12-29:
I too would love a dark theme. Another very similar software has it but that one would never "remember" where I put it and I would have to re-position it every time it started up. This one stays put but the white background is rough on my old eyes.

LoDef, 2022-01-28:
+1 for dark theme (or "use system theme").

Damians0, 2023-04-20:
Is the dark theme still not available for this program?

Ryder17z, 2024-07-17:
> Is the dark theme still not available for this program?

Apparently not.