openhardwaremonitor/openhardwaremonitor #1386: NVIDIA GPU Memory Stats Misreporting

tomgade09, 2021-01-06:
I have two NVIDIA GPUs in my system - a GTX 1080 connected via Thunderbolt 3 to an external enclosure (Razer Core X), and an internal GTX 960M.  It appears as if the memory size and load is reporting incorrectly for the 960M.  The 960M is reported as having 8GB of memory, when it only has 2GB, and the load on the memory controller of the 960M is exactly the same as that of 1080.  If you're using CUDA, it could maybe be something as simple as not using a cudaSetDevice prior to polling.  Clock speed appears to be reported correctly.

I'm on the latest version offered by your website (not github) - 0.9.6.

Great product otherwise!

msawyer91, 2022-04-02:
I have the exact same problem, and it appears in all six of my desktops. Just looking at one of them, I have an RTX 3060 with 12GB of RAM and an RTX 2060 with 6GB of RAM. This is confirmed by Windows Task Manager. However, when running Open Hardware Monitor (both 0.9.6 and 0.10.0.1), the memory statistics are identical for both GPUs -- it appears that since the RTX 3060 is GPU 0 and the RTX 2060 is GPU 1, the RTX 3060's memory details are duplicated. All of the other stats, such as power consumption, fan speed, clocks (including memory speed), temperature, and throughput all seem to be reflected correctly for each card. It's just the three memory stats in the Data section -- memory free, used, total -- and GPU Memory under Load, which are incorrect on the second GPU. I have a desktop with three CPUs and the same behavior is observed -- the memory stats of the first GPU are reflected across the second and third GPUs, while all other stats are correct.

Below are screenshots of Open Hardware Monitor (built from Git source in VS2022 Community) and Task Manager of one of my desktops, showing the RTX 3060 and 2060.

I'm not sure if this matters, but only the first GPU is plugged directly into the mobo. All other GPUs are attached to riser cards as they are used for mining. I wouldn't think this should matter, since all other statistics are reported correctly.

[image]
[image]
[image]

msawyer91, 2022-04-02:
I poked around in the debugger a bit, and will probably play around with this a bit more later. But one thing I found interesting was the displayHandle value. For the RTX 3060, the displayHandle.ptr value is 0x00000000de000001 but for the RTX 2060, displayHandle.ptr value is 0x0000000000000000. Either way, the "if" statement returns true and the data is updated, despite the 0x0000000000000000 value. It may be a red herring, or it may be an incorrect handle, hence the misrepresented memory stats.

RTX 3060:
[image]

RTX 2060:
[image]

msawyer91, 2022-04-02:
> I have two NVIDIA GPUs in my system - a GTX 1080 connected via Thunderbolt 3 to an external enclosure (Razer Core X), and an internal GTX 960M. It appears as if the memory size and load is reporting incorrectly for the 960M. The 960M is reported as having 8GB of memory, when it only has 2GB, and the load on the memory controller of the 960M is exactly the same as that of 1080. If you're using CUDA, it could maybe be something as simple as not using a cudaSetDevice prior to polling. Clock speed appears to be reported correctly.
> 
> On my system, CUDA reports the GTX 1080 as device 0, and the 960M as device 1.
> 
> I'm on the latest version offered by your website (not github) - 0.9.6.
> 
> Great product otherwise!

I dug into this and put together a fix. Please check out my [pull request](https://github.com/hexagon-oss/openhardwaremonitor/pull/28). You can also grab a [release here](https://github.com/msawyer91/openhardwaremonitor/releases/tag/v0.10.1-alpha3).