openhardwaremonitor/openhardwaremonitor #1387: Feature request: Is there an OBS plugin to see OHM hardware graph overlays?

MBB232, 2021-01-11:
I suppose it is possible to capture the graph window and then remove the background with a color filter. 
But I suspect it would be less hardware intensive to have it drawn in OBS itself. 

So I was wondering if there is a plugin that allows the projecting of the OHM stats as stream graph(s) in OBS studio? (Windows in my case, but crossplatform would be the coolest. )
It may be useful during recording of a bug/testing for a software to see its hardware use, and possibly overlay it to a recording of the same action in a previous version.

lprocx, 2021-01-28:
I don't know if this is possible directly with OBS, but depending on your own skills you could load OHM into an application and program it yourself. Personally I'm working on something like this (display the data, but in my own way). I'm at the stage where I can read all the data and store it. Exposing it through an REST API, but any option from this point is duable.

shrddr, 2021-07-11:
You can Log sensors to a file and then render that with some generic graph-drawing overlay