openhardwaremonitor/openhardwaremonitor #1391: MSI MEG Z490 ACE (MS-7c71) shows nothing!

TorstenAden, 2021-01-22:
hi I can not see any info at all from my motherboard (MSI MEG Z490 ACE (MS-7c71)), can this be fixed?

I have version 0.9.6 Open Hardware Monitor

Hope someone can fix this
//Torsten

[https://drive.google.com/file/d/1mg4WkYDw4g4elvfFpuBB-Y3DtjWnwAI0/view?usp=sharing](url)

kleuter, 2021-01-26:
Here's the picture actually
[image]

tippee, 2021-01-28:
I have the same problem with a similar z490 msi motherboard. im on the latest bios version and running 0.9.6 OHM0. the problem occurs in 0.9.5 also.

[image]

kleuter, 2021-01-28:
You probably better post report than screenshots.