openhardwaremonitor/openhardwaremonitor #1393: MCP temperature

andreas-R17, 2021-01-31:
I have a motherboard on nForce 750a (MSI MS-7578). Why Open Hardware Monitor does not show the temperature of MCP?

[image]

shrddr, 2021-07-11:
What is MCP is and why is it important to you?

andreas-R17, 2021-07-12:
MCP -  Media Communications Processor and incorporates both Northbridge and Southbridge on a single die.
Сhipset is very hot and requires temperature control.

shrddr, 2021-07-12:
Every motherboard has a hardware monitor chip which is a common way to gather and manage temp, voltage and fan data. Your F71889F indeed has 3 temperature inputs (check the datasheet) and that's how much temp sensors you get in OHM. These sensors are initially unnamed though because the software doesn't know what devices are physically connected to those 3 chip inputs. You can make sense which one means what by a) comparing values in OHM and other hardware monitoring software, preferably by your motherboard vendor b) physically heating and cooling down your motherboard components and look how OHM values react. After that you can rename the sensors using context menu.
If you check all three sensors and neither one is MCP, I don't know... maybe contact your vendor and ask what is the correct way to get MCP temperature from software

andreas-R17, 2021-07-12:
The first sensor is the CPU, the second sensor is the motherboard temperature, the third sensor is probably not connected.
But some programs show MCP. OCCT is an overclocking test program.
It is not convenient to monitor the temperature in this program.
Unfortunately I don't know how they get the temperature.
Аnd this program doesn't work on Linux((.

[image]

shrddr, 2021-07-12:
With 2 decimals precision, the value is certainly not coming from F71889F because this chip only has 8 bit ADC. So your motherboard must have an additional interface specifically for monitoring the MCP. If you're feeling adventurous you can try to disassemble the OCCT binary to learn more how it accesses the data.

andreas-R17, 2021-07-13:
Thanks for trying to help.