openhardwaremonitor/openhardwaremonitor #1394: B550 Gaming Edge not showing sensors

SA2178, 2021-02-02:
I have a MSI B550 Gaming Edge WIFI which is not showing any of its sensors. I am running version 0.9.6. While I have hidden some sensor, I did not hide any mainboard sensors. Attached I have a screenshot, please let me know if there is anything else I can provide.
[image]

KanyeMiddleEast, 2021-02-05:
[HWMonitorPRO.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5931833/HWMonitorPRO.txt)

DarthDestroyer, 2021-02-05:
It would be great if you could post a ohm report instead of third party data reports!

KanyeMiddleEast, 2021-02-05:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5931912/OpenHardwareMonitor.Report.txt)

KanyeMiddleEast, 2021-02-05:
> It would be great if you could post a ohm report instead of third party data reports!

Sorry about that, I was looking to back up my claim that my board uses the nct6687, and ohm didn't show much information about my board

DarthDestroyer, 2021-02-05:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD592 at 0x4E/0x4F

i have seen this id in another issue!

KanyeMiddleEast, 2021-02-05:
> Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD592 at 0x4E/0x4F
> 
> i have seen this id in another issue!

I believe that may be because the nuvoton nct6687 does not have compatibility currently, so I believe that may be what it says when the chip isn't supported

gchennell, 2021-02-21:
> Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD592 at 0x4E/0x4F
> 
> i have seen this id in another issue!

> > Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD592 at 0x4E/0x4F
> > i have seen this id in another issue!
> 
> I believe that may be because the nuvoton nct6687 does not have compatibility currently, so I believe that may be what it says when the chip isn't supported

Having same issue with the b550 Carbon WIFI - probably same i/o chip but I can't see how to find that.
Would be great to have resolved - I'm trying to get a script to poll sensors/temps to send to a remote display - or get them on the web server to then be redisplayed on said screen (on a Pi probably = not having a third screen ran by my GPU) I will be sharing code once this project is completed. Hoping it works out well

KanyeMiddleEast, 2021-02-21:
> Having same issue with the b550 Carbon WIFI - probably same i/o chip but I can't see how to find that.
> Would be great to have resolved - I'm trying to get a script to poll sensors/temps to send to a [...]

It is the same chip. I checked by looking up my motherboard on MSI's website, going to specifications, clicking the details tab, and the I/O Controller is listed there. I tried to get someone from Fivrr to write the code and put in a pull request, but that didn't work out, unfortunately. Please do share the code if you figure it out!

UhtredTheBold, 2021-06-23:
I see the same issue with my MSI MPG B550 GAMING EDGE WIFI (MS-7C91)

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD592 at 0x4E/0x4F

Iciclemelt, 2022-01-04:
Me too, I see the same issue with my MSI MPG B550 GAMING PLUS

gchennell, 2022-01-04:
I have discovered LibreHardwareMonitor. It seems to do what I want and I
have started making some progress towards my project now.

Iciclemelt, 2022-01-04:
> LibreHardwareMonitor

> I have discovered LibreHardwareMonitor. It seems to do what I want and I have started making some progress towards my project now.
> […](#)

Ok thanks but i use OpenHardwareMonitor with Rainmeter with the "OpenHardwareMonitorPlugin.dll" and this one not exist for LibreHardwareMonitor ?