openhardwaremonitor/openhardwaremonitor #1398: SSD wear level not measured

Bullion-Manor, 2021-02-08:
I use Windows 10. My device has an SSD for storage, but this program detects it as a "Generic Hard Disk". It measured my storage usage correctly within what seems to be 2% error according to the data from Disk Management. My question now is if SSDs are supposed to be detected as a "General Hard Disk", and whether or not this is an actual bug. Regardless, an issue persists because the [official website's](https://openhardwaremonitor.org/documentation/) documentation page said that SSD wear level is supposed to be displayed, but it isn't.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/5941368/OpenHardwareMonitor.Report.txt)

evg656e, 2021-02-12:
Looks like it does not recognize NVMe drives
[image]

donpin1, 2021-02-15:
Agree, please add NVMe drives and their temperatures.  It would be great if you also had a tab that monitored SMART attributes.   Nevertheless, this is easily the best program for me that I have come across. Well done.
[image]

Scott-G, 2021-02-23:
Hard agree that NVMe support is needed but, even without it, this is still the best hardware monitor, by far.